ZoneMinder/zoneminder #2917: Error in logs regarding filter SQL format

XeonNAS, 2020-04-21:
**Describe Your Environment**
- Version of ZoneMinder: 1.34.9
- How you installed ZoneMinder: PPA
- Ubuntu Bionic, headless

**Describe the bug**
Noticed Log was marked as red(error), entered logs to see every filter that is used to purge old events is throwing an error:
Can't execute filter 'SELECT E.*, unix_timestamp(E.StartTime) as Time, M.Name as MonitorName, M.DefaultRate, M.DefaultScale FROM Events as E INNER JOIN Monitors as M on M.Id = E.MonitorId LEFT JOIN Storage as S on S.Id = E.StorageId WHERE (E.EndTime IS NOT NULL) AND ( M.Name = 'DoorBell' and E.AlarmFrames >= -10 day ) ORDER BY E.Id DESC', ignoring: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'day ) ORDER BY E.Id DESC' at line 9

**To Reproduce**
Steps to reproduce the behavior:
1. Go to console in browser bar
2. Click on Logs, set level to Error
3. See error

**Expected behavior**
No zmfilter.pl errors

**Debug Logs**
```

04/21/20 11:19:12.508471 zmfilter_8[2889].DBG [ZoneMinder::Logger:294] [LogOpts: level=DBG/DBG, screen=OFF, database=ERR, logfile=DBG->/var/log/zm/zm_debug.log.2889, syslog=DBG]
04/21/20 11:19:12.508830 zmfilter_8[2889].INF [main:180] [Scanning for events using filter id '8']
04/21/20 11:19:12.508947 zmfilter_8[2889].DBG [main:199] [Reloading filters]
04/21/20 11:19:12.509832 zmfilter_8[2889].DBG [main:255] [Found filter 'Delete 10 Day Old DoorBell']
04/21/20 11:19:12.516956 zmfilter_8[2889].DBG [main:269] [Got 1 filters]
04/21/20 11:19:12.517473 zmfilter_8[2889].DBG [ZoneMinder::Filter:113] [Filter::Execute SQL (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( M.Name = 'DoorBell' and E.AlarmFrames >= -10 day ) ORDER BY E.Id DESC)]
04/21/20 11:19:12.517910 zmfilter_8[2889].ERR [ZoneMinder::Filter:118] [Can't execute filter 'SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( M.Name = 'DoorBell' and E.AlarmFrames >= -10 day ) ORDER BY E.Id DESC', ignoring: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'day ) ORDER BY E.Id DESC' at line 9]
04/21/20 11:19:12.561768 zmfilter_8[2889].DBG [main:292] [Checking filter Delete 10 Day Old DoorBell delete returned 0 events ]

```

connortechnology, 2020-04-21:
Your filter is badly formed.  Note in the SQL the E.AlarmFrames >= -10 day.  So the line in your filter that SHOULD be StartDate/Time has been changed to AlarmFrames.  Also please note that >= -10 days means NEWER than -10 days.  You want it to be less than.  Older times are a smaller value than newer times.

closed 2020-04-21 by connortechnology

XeonNAS, 2020-04-21:
Interesting, this filter has been in place for years and worked OK in the past. I'll have a look at fixing it.

John

Closed #2917<https://github.com/ZoneMinder/zoneminder/issues/2917>.