ZoneMinder/zoneminder #2918: Mootools breaks google recaptcha

PVi1, 2020-04-21:
**Describe Your Environment**
- Version of ZoneMinder - 1.43
- How you installed ZoneMinder - PPA http://ppa.launchpad.net/iconnor/zoneminder-1.34/ubuntu bionic
- Full name and version of OS - 	Ubuntu 18.04.4 LTS 64bit
- Browser name and version (if this is an issue with the web interface) - Mozilla Firefox 75.0
- google Recaptcha v2 release: https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js (other language versions are the same)

**Describe the bug**
Mootools breaks google recaptcha. 
Loading of the login page ends with the following javascript error in browser console:
```
TypeError: g is undefined recaptcha__sk.js:235:289
bL https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:235
    bind https://example.org/tools/mootools/mootools-core.js:1
    tW https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:298
    bC https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:159
    YI https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:163
    dA https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:243
    Z4 https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:613
    cH https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:182
    I https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:225
    forEach self-hosted:235
    $I https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:225
    wY https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:250
    CA https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:296
    h https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:336
    CA https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:296
    <anonymous> https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:614
    <anonymous> https://www.gstatic.com/recaptcha/releases/wk6lx42JIeYmEAQSHndnyT8Q/recaptcha__sk.js:627
```

**To Reproduce**
Steps to reproduce the behavior:
Enable&configure google recaptcha. Try to login. Recaptcha will not load and you are unable to login.

**Expected behavior**
Recaptcha should load, and user should be able to login.

connortechnology, 2020-04-21:
Confirmed.  I have a fix though.  Will release 1.34.10 soon with it.

phuongpham, 2020-04-23:
I have same issue. @connortechnology  do you know when is the next release with this fix?

connortechnology, 2020-04-23:
likely today

phuongpham, 2020-04-24:
@connortechnology I just installed the latest build. But now I received this error:

ERR | reCaptcha authentication failed. No g-recpatcha-response in REQUEST: | /usr/share/zoneminder/www/includes/actions/login.php | 35

I'm not sure if this error is releated to the fix that is done.

connortechnology, 2020-04-24:
Dang.  I swear it was working for me yesterday.

Today I have a patch that will simple not include mootools on the login page.  This appears to make it work again.

Please see commit cf3785aebf13c3d1949a61a4a94064f12106c9ad

phuongpham, 2020-04-24:
Great! I just updated the functions.php file manually and can confirms it works now! Thanks for your help!

lstark14, 2020-04-25:
Thanks!! This was very helpful. I am able to login. You guys rock!

commit 3e8de7dd (2020-04-22): Move the script include for recaptcha into the view instead of the xhtmlHeaders function.  Fixes #2918
closed 2020-05-13 by connortechnology