ZoneMinder/zoneminder #2923: Blank Window when Trying to View Events

RebornJumpman, 2020-04-24:
**Describe Your Environment**
- Version 1.32.3
- Installed by RPMFusion
- Running on CentOS 8
- Attempted using Firefox, Chrome, Edge, Vivaldi, and Brave

**If the issue concerns a camera**
- The cameras do not appear to be the problem

**Describe the bug**
Monitors appear, but when trying to view an event, a blank window opens and nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console'
2. Click on any monitor
3. Click on any event
4. See blank window

**Expected behavior**
The event player should show the video of the event in question as it has done in previous versions.

connortechnology, 2020-04-24:
1.32 is old.  Please update.

intc, 2020-05-23:
We have one fresh installation with similar issues. Live views work fine. Events play **if** scale=actual - Anything else than that will cause black screen and nasty looking errors in the log:
```
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Got signal 6 (Aborted), crashing]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Signal address is 0x510000425c, from 0x7f4a2aa47901]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 0: /var/www/localhost/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x76) [0x55d972aa4036]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 1: /lib64/libpthread.so.0(+0x12fe0) [0x7f4a2d142fe0]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 2: /lib64/libc.so.6(gsignal+0x141) [0x7f4a2aa47901]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 3: /lib64/libc.so.6(abort+0x127) [0x7f4a2aa312dd]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 4: /lib64/libc.so.6(+0x7a2ce) [0x7f4a2aa8a2ce]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 5: /lib64/libc.so.6(+0x81264) [0x7f4a2aa91264]]
May 22 19:15:50 zm zms_e73[16988]: ERR [zms_e73] [Backtrace 6: /lib64/libc.so.6(+0x82bd4) [0x7f4a2aa92bd4]]
... ...
May 23 18:43:04 zm zms_e13[9428]: ERR [zms_e13] [Got signal 11 (Segmentation fault), crashing]
May 23 18:43:04 zm zms_e13[9428]: ERR [zms_e13] [Signal address is 0x7fdb76436038, from 0x7fdb7915ee70]
May 23 18:43:04 zm zms_e13[9428]: ERR [zms_e13] [Backtrace 0: /var/www/localhost/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x76) [0x55f56acad036]]
May 23 18:43:04 zm zms_e13[9428]: ERR [zms_e13] [Backtrace 1: /lib64/libpthread.so.0(+0x12fe0) [0x7fdb7b80bfe0]]
May 23 18:43:04 zm zms_e13[9428]: ERR [zms_e13] [Backtrace 2: /lib64/libc.so.6(cfree+0x18) [0x7fdb7915ee70]]
```
ZM was built against 46a13237b52f1835d2e985d4b3b2f19f28e258e8.

This runs on Gentoo. 
```
sys-devel/gcc-9.3.0
sys-libs/glibc-2.30-r8
media-video/ffmpeg-4.2.2
dev-lang/perl-5.30.1
```

In addition - if scale=actual - and if one let's the event play till it's end - we'll see a long stream of log lines like these:
```
May 23 18:51:38 zm zms_e13[9522]: ERR [zms_e13] [Can't open /var/lib/zoneminder/events/3/2020-05-20/13/01884-capture.jpg: No such file or directory]
May 23 18:51:38 zm zms_e13[9522]: ERR [zms_e13] [Can't open /var/lib/zoneminder/events/3/2020-05-20/13/01885-capture.jpg: No such file or directory]
May 23 18:51:38 zm zms_e13[9522]: ERR [zms_e13] [Can't open /var/lib/zoneminder/events/3/2020-05-20/13/01886-capture.jpg: No such file or directory]
May 23 18:51:38 zm zms_e13[9522]: ERR [zms_e13] [Can't open /var/lib/zoneminder/events/3/2020-05-20/13/01887-capture.jpg: No such file or directory]
```
It seems that zms_e?? continues to iterate for some reason even there are no frames left. If one stops the playback well before the end of the stream ends these errors will not appear.

closed 2020-05-23 by connortechnology

intc, 2020-05-24:
Thank you for fixing the scaling issue!

commit 662b9894 (2020-05-23): Don't delete send_image because it is static.  Fixes #2923