ZoneMinder/zoneminder #2925: 'zma -m 2' exited abnormally, exit status 255

arthur798, 2020-04-28:
Hiya I have this config, i know this is not exactly zoneminder official image but it's close enough

  zoneminder:
    image: quantumobject/docker-zoneminder:1.34
#    image: adc3c7a0ca0a
    container_name: zoneminder
    restart: unless-stopped
    networks:
      - t2_proxy
    security_opt:
      - no-new-privileges:true
    ports:
      - "$ZONEMINDER_HTTP_PORT:80"
#      - "$ZONEMINDER_HTTPS_PORT:443"
      - "9567:9000" # Event Notification Port
    privileged: true
    shm_size: 4096m
    volumes:
      - /var/empty
      - $USERDIR/docker/shared:/shared
      - $USERDIR/docker/zoneminder/backups:/var/backups
      - $USERDIR/docker/zoneminder:/var/cache/zoneminder
      - $USERDIR/docker/zoneminder/config:/config
      - type: tmpfs
        target: /dev/shm
    environment:
      SHMEM: "50%"
      PUID: $PUID
      PGID: $PGID
      TZ: $TZ
      PHP_TZ: $TZ
      ZM_DB_HOST: $DB_HOST
      ZM_DB_PORT: $DB_PORT
      MYSQL_ROOT_PASSWORD: $MYSQL_ROOT_PASSWORD
      MYSQL_ROOT: $USER
also I am getting

Got empty memory map file size 0, is the zmc process for this monitor running?

what could be the issue?

connortechnology, 2020-05-02:
zma is looking for the shared mem files created by zmc.  So the question is why is zmc not creating them?

arthur798, 2020-05-02:
What can I provide to help debug this?

connortechnology, 2020-05-02:
You could turn on debugging and look in your zmc logs.

arthur798, 2020-05-02:
What's the best/easiest way to do that?

SteveGilvarry, 2020-05-02:
https://zoneminder.readthedocs.io/en/stable/userguide/options/options_logging.html

Try having a quick look here and you will want _zmc_m2 for the target from memory, but the docs section on debug target will confirm.

arthur798, 2020-05-03:
Thanks will take a look and come back with some info

arthur798, 2020-05-04:
```
05/04/20 09:23:27.401082 zmc_m2[277422].INF-zm_ffmpeg.cpp/84 [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor is not part of your debug targets]
05/04/20 09:23:27.678565 zmc_m2[277422].INF-zmc.cpp/223 [Starting Capture version 1.34.9]
05/04/20 09:23:27.682000 zmc_m2[277422].INF-zm_ffmpeg_camera.cpp/201 [Priming capture from rtsp://admin:dg20160404@192.168.1.164/onvif1]
05/04/20 09:23:38.472041 zmc_m2[277422].ERR-zm_ffmpeg_camera.cpp/379 [Unable to open input rtsp://admin:dg20160404@192.168.1.164/onvif1 due to: Invalid data found when processing input]
05/04/20 09:23:50.152500 zmc_m2[277435].INF-zm_ffmpeg.cpp/84 [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor is not part of your debug targets]
05/04/20 09:23:50.672632 zmc_m2[277435].INF-zmc.cpp/223 [Starting Capture version 1.34.9]
05/04/20 09:23:50.675272 zmc_m2[277435].INF-zm_ffmpeg_camera.cpp/201 [Priming capture from rtsp://admin:dg20160404@192.168.1.164/onvif1]
05/04/20 09:23:50.845728 zmc_m2[277435].ERR-zm_ffmpeg_camera.cpp/379 [Unable to open input rtsp://admin:dg20160404@192.168.1.164/onvif1 due to: Interrupted system call]
```

what does that mean? that's coming from the zmc_m2.log file

connortechnology, 2020-05-04:
Looks like it is being killed before it can successfully open the stream.  Some cameras take a long time.  Check your value for 
WATCH_MAX_DELAY in Options -> SYSTEM.  Make it 45 or more.

arthur798, 2020-05-04:
I changed it to 50 seconds, still happening do I need to restart zoneminder or something

connortechnology, 2020-05-15:
The /onvif1 is a bit suspicious.  Looks like you have the rtsp url wrong.  

Try it in vlc and see if it works there.

arthur798, 2020-05-15:
It's definitely fine as sometimes it's recording

vvchumanov, 2020-06-06:
`
2020-06-07 00:45:16 | zms_m7 |   | 11171 | ERR | Unable to authenticate user | zms.cpp | 207
-- | -- | -- | -- | -- | -- | -- | --
`
what this after update to 1.34.15, al zms_mXX..

closed 2021-03-10 by stale[bot]