ZoneMinder/zoneminder #2927: HTTP 500 when trying to add storage on 1.35.1

stack-fault, 2020-05-01:
Hi, I am getting an HTTP 500 when trying to add storage.

When looking at the apache logs, here is the line describing the issue:

```[Fri May 01 11:25:43.856158 2020] [php7:error] [pid 27119] [client {ip-redacted}:55642] PHP Fatal error:  Uncaught Error: Class 'Storage' not found in /usr/share/zoneminder/www/skins/classic/views/storage.php:32\nStack trace:\n#0 /usr/share/zoneminder/www/index.php(284): require_once()\n#1 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/views/storage.php on line 32, referer: http://{site-redacted}/zm/?view=options&tab=storage```

I am currently using 1.35.1 on Debian Buster 10. Installed from the repo at hxxp://zmrepo.zoneminder.com/debian/master buster/

Thanks

connortechnology, 2020-05-01:
1.35.1 is old.  This was already fixed.  Please update to 1.35.2.

stack-fault, 2020-05-01:
Looked at the repo manually and had to give it a nudge to finally figure out there was an update and take it. Funny thing, ZM reported 1.35.1 as being the latest when clicking on the version at the top. Sorry for that.

closed 2020-05-01 by stack-fault