ZoneMinder/zoneminder #2929: Hour Events aren't cleared

CyberGuerro, 2020-05-07:
Environment:
- ZoneMinder 1.34.11 with MySQL
- Linux installation from source
- Linux Gentoo stable
- Firefox

If more then a hour is passed since event is registered, that event isn't cleared from "Hour Events" (resetting fields HourEvents and HourEventDiskspace).
If more than a day / week / month is passed, event is cleared from corresponding fields.

It is always reproducible

CyberGuerro, 2020-05-07:
I found that during an upgrade, trigger **Events_Hour_delete_trigger** wasn't created.
Since I created it, all works again.

closed 2020-05-07 by CyberGuerro

connortechnology, 2020-05-07:
I guess the issue here is "Why didn't the trigger get created?!" What version were you upgrading from/to?

CyberGuerro, 2020-05-07:
I have this issue since a couple of month, I belive  about from 1.34.4 to 1.34.5, but I'm not sure... sorry