ZoneMinder/zoneminder #2930: Mysql 8 Error: "Function" is now a reserved word

Elwinmage, 2020-05-07:
Ubuntu 19.10
Mysql 8.0.20
Zoneminder 1.34

**Describe the bug**
Since Mysql 8.0.1 "Function" is now a reserved word.
Zone Minder use Function as a column name in Monitors Table.

**To Reproduce**
Steps to reproduce the behavior:
1. mysql -p
2. use zm
3. select Function from Monitors;
4. ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function from Monitors' at line 1

**Expected behavior**
Change the name of the column in Monitors Table.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2020-05-14:
put it in ``'s.  SELECT `Function` From Monitors;

marillat, 2020-09-25:
@connortechnology  why this bug has been closed ?
This issue has been reported in Debian and Ubuntu with patch
https://bugs.debian.org/970906
https://bugs.launchpad.net/ubuntu/+source/zoneminder/+bug/1859295

SteveGilvarry, 2020-09-25:
In the issue you linked says "I've backported a patch to 1.32.3 from upstream git master." This means @connortechnology fixed it and they copied it.  This is the commit that fixed the issue https://github.com/ZoneMinder/zoneminder/commit/b1b258ab9b26c6813fe65d5ee432dccfb2e9051a

Your package may not have been updated, current release is 1.34.21

marillat, 2020-09-25:
I'm using 1.34.21
Apparently you have missed to quote some Function keyword as displayed  in this patch https://bugs.debian.org/cgi-bin/bugreport.cgi?att=1;bug=970906;filename=zoneminder_1.34.16-2ubuntu1.patch;msg=5

SteveGilvarry, 2020-09-25:
OK so the Patch is identical to master. But those changes have not been pushed back to a 1.34 release. https://github.com/ZoneMinder/zoneminder/commit/cc93d553d88d352e7c2f77adb246fd4154a4c804

Also looks like both the patch and master are missing `` on the second usage of function in line 153 and 155
https://github.com/ZoneMinder/zoneminder/blob/bcb430d3937e92007bc11d7e56f020b2f973ec14/scripts/ZoneMinder/lib/ZoneMinder/Database.pm#L153.
```
      $sql .= " where `Function` = 'Modect' or Function = 'Mocord'";
```
Should be
```
      $sql .= " where `Function` = 'Modect' or `Function` = 'Mocord'";
```

marillat, 2020-09-25:
> OK so the Patch is identical to master. But those changes have not been pushed back to a 1.34 release. https://github.com/ZoneMinder/zoneminder/commit/cc93d553d88d352e7c2f77adb246fd4154a4c804

It is possible to push commit cc93d55 to 1.34 release branch so Debian/Ubuntu people (and others) will be happy ?

connortechnology, 2020-09-25:
I'll tag it for 1.34.22

marillat, 2020-09-25:
@connortechnology  Thanks for the push in 1.34

connortechnology, 2020-10-19:
Closing as I have just released 1.34.22 with the fixes.

closed 2020-10-19 by connortechnology