ZoneMinder/zoneminder #2931: DCS5020L.pm not running

stevaedrum, 2020-05-09:
-  ZoneMinder 1.34.11
- installed with RPMFusion
- Fedora Core 31
- Chrome & Firefox

- D-Link DCS5020L

PTZ not running and exit error on zmcontrol

I have detected in the DCS5020L.pm un bug in the version and master.
This sub is missing "sub printMsg", present in 1.34.9 version.

I have add this and working normaly.

sub printMsg
{
    my $self = shift;
    my $msg = shift;
    my $msg_len = length($msg);

    Debug( $msg."[".$msg_len."]" );
}

connortechnology, 2020-05-09:
The better thing to do is 
Line 72 of /usr/share/perl5/ZoneMinder/Control/DCS5020L.pm should be:
$self->printMsg($cmd, 'Tx');

stevaedrum, 2020-05-09:
Thanks for your feedback, I have made your proposal, it's working.
But don't correct issue #2932

stevaedrum, 2020-05-21:
Hi all,

I permit to share a problem in branch
Concerning this issue, I have detected that in master branch it's correct but not in branch 1.34.15, 1.34.14, 1.34.13....

 $self->printMsg($cmd, 'Tx');

Thnaks in advance for your support.

connortechnology, 2020-06-15:
fixed in 1.34.16

closed 2020-06-15 by connortechnology