ZoneMinder/zoneminder #2932: PTZ arrow in monitor not working

stevaedrum, 2020-05-09:
ZoneMinder 1.34.11
installed with RPMFusion
Fedora Core 31
Chrome & Firefox

D-Link DCS5020L

"PantiltButtons" on the page of monitor not running with Chrome or Firefox, I click on arrow button no move.
PresetButton, Wake, Sleep, Reset and PresetHome working correctly.

I don't have error message.

It's running with ZmNija, with API.

connortechnology, 2020-05-09:
There is a known issue with DCS5020L in 1.34.11.  Line 72 of /usr/share/perl5/ZoneMinder/Control/DCS5020L.pm should be:
$self->printMsg($cmd, 'Tx');
Can you change that line and report if it fixes your problem?

stevaedrum, 2020-05-09:
This issue is not corrected,
"PantiltButtons" on the page of monitor not running

closed 2020-05-13 by connortechnology

connortechnology, 2020-05-13:
I will release 1.34.13 with this fix in a few days.

commit 807efde5 (2020-05-13): xtell and ytell are data-xtell and data-ytell fixes #2932
commit edd718af (2020-05-13): xtell and ytell are data-xtell and data-ytell fixes #2932