ZoneMinder/zoneminder #2935: jpeg errors

mad1524, 2020-05-13:
- Version of ZoneMinder 1.34
- name and version of OS - ubuntu server 18.04
I tried to connect IP cameras - it did not work, there were some strange errors.
then I put the server and Zoneminder from scratch. I started to add trying on wifi android and iphone.
The iPhone shows in green, but a very bad image, blurry and almost invisible.
an old android does not connect at all (although it worked fine on the local server and the image quality was not bad).
The logs are as follows:

held buffer is undersized for requested buffer
failed requesting writeable buffer for reading JPEG image
unable to decode jpeg

I came across this program for the first time, but it is very necessary to configure everything qualitatively. Help me please.

[image]

connortechnology, 2020-05-13:
Sounds like you have not correctly specified the size of the input stream, and perhaps the # of colours on the Source tab of the monitor settings.

This is not a bug.  You have not read the rules about posting here.
For free support go to https://forums.zoneminder.com

closed 2020-05-13 by connortechnology

mad1524, 2020-05-13:
Nowhere else to turn.
The data indicated correctly (specifically for the camera). She worked on a home local server.