ZoneMinder/zoneminder #2936: Zoneminder building failed

zhuykovkb, 2020-05-14:
**Describe Your Environment**
- Version of ZoneMinder: commit 5f5cede104a6a6c4b67480649392d1a61a4780a2
- How you installed ZoneMinder: build from source using https://github.com/ZoneMinder/zmdockerfiles/blob/master/development/ubuntu/xenial/Dockerfile
- OS: Linux Ubuntu xenial in image
- no browser needed

**Describe the bug**
Build of zoneminder failed. See details in attachment log file

**To Reproduce**
Steps to reproduce the behavior:
1. clone repo https://github.com/ZoneMinder/zoneminder
2. try to build it

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```
Step 7/19 : RUN make && make install
 ---> Running in 8f0a5d6f6cd7
Scanning dependencies of target zm
[  1%] Building CXX object src/CMakeFiles/zm.dir/zm_box.cpp.o
[  3%] Building CXX object src/CMakeFiles/zm.dir/zm_buffer.cpp.o
[  3%] Building CXX object src/CMakeFiles/zm.dir/zm_camera.cpp.o
[  4%] Building CXX object src/CMakeFiles/zm.dir/zm_comms.cpp.o
[  6%] Building CXX object src/CMakeFiles/zm.dir/zm_config.cpp.o
[  7%] Building CXX object src/CMakeFiles/zm.dir/zm_coord.cpp.o
[  9%] Building CXX object src/CMakeFiles/zm.dir/zm_curl_camera.cpp.o
[ 11%] Building CXX object src/CMakeFiles/zm.dir/zm.cpp.o
[ 12%] Building CXX object src/CMakeFiles/zm.dir/zm_db.cpp.o
[ 12%] Building CXX object src/CMakeFiles/zm.dir/zm_logger.cpp.o
[ 14%] Building CXX object src/CMakeFiles/zm.dir/zm_event.cpp.o
[91m/ZoneMinder/src/zm_event.cpp: In member function 'void Event::AddFrame(Image*, timeval, int, Image*)':
/ZoneMinder/src/zm_event.cpp:676:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( write_to_db || (frame_data.size() > (int)monitor->get_fps()) ) {
                                            ^
[0m[ 15%] Building CXX object src/CMakeFiles/zm.dir/zm_frame.cpp.o
[ 17%] Building CXX object src/CMakeFiles/zm.dir/zm_eventstream.cpp.o
[91m/ZoneMinder/src/zm_eventstream.cpp: In member function 'bool EventStream::loadEventData(uint64_t)':
/ZoneMinder/src/zm_eventstream.cpp:169:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   } else if ( monitor->Id() != event_data->monitor_id ) {
                             ^
[0m[91m/ZoneMinder/src/zm_eventstream.cpp: In member function 'virtual void EventStream::runStream()':
/ZoneMinder/src/zm_eventstream.cpp:869:8: warning: unused variable 'in_event' [-Wunused-variable]
   bool in_event = true;
        ^
[0m[ 19%] Building CXX object src/CMakeFiles/zm.dir/zm_exception.cpp.o
[ 20%] Building CXX object src/CMakeFiles/zm.dir/zm_file_camera.cpp.o
[ 20%] Building CXX object src/CMakeFiles/zm.dir/zm_ffmpeg_input.cpp.o
[ 22%] Building CXX object src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o
[91m/ZoneMinder/src/zm_ffmpeg_camera.cpp: In member function 'virtual int FfmpegCamera::CaptureAndRecord(Image&, timeval, char*)':
/ZoneMinder/src/zm_ffmpeg_camera.cpp:978:50: error: 'hw_pix_fmt' was not declared in this scope
           if ( (ret == AVERROR_INVALIDDATA ) && (hw_pix_fmt != AV_PIX_FMT_NONE) ) {
                                                  ^
[0m[91mmake[2]: *** [src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o] Error 1
[0msrc/CMakeFiles/zm.dir/build.make:446: recipe for target 'src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o' failed
CMakeFiles/Makefile2:376: recipe for target 'src/CMakeFiles/zm.dir/all' failed
[91mmake[1]: *** [src/CMakeFiles/zm.dir/all] Error 2
make: *** [all] Error 2
[0mMakefile:149: recipe for target 'all' failed
Service 'zm' failed to build: The command '/bin/sh -c make && make install' returned a non-zero code: 2

```
[issue.log](https://github.com/ZoneMinder/zoneminder/files/4626981/issue.log)

zhuykovkb, 2020-05-14:
@connortechnology please, help me. Tell me, if you need more info :)

connortechnology, 2020-05-14:
fix applied.

zhuykovkb, 2020-05-14:
thank you @connortechnology , it works =)

closed 2020-05-14 by zhuykovkb