ZoneMinder/zoneminder #2937: zmu failure causes API call to hang and orphans a database connection

mhilbush, 2020-05-14:
- Version of ZoneMinder: 1.34.12
- How you installed ZoneMinder: unknown
- Full name and version of OS: unknown

**Describe the bug**
API call hangs and orphans a database connection

[My app](https://github.com/openhab/openhab-addons/pull/7412) calls the following API to determine if a monitor is alarming.
`http://zoneminder/zm/api/monitors/alarm/id:8/command:status.json`

A user of my app recently reported that his Zoneminder installation was running out of database connections over time.

I had the user execute the above API call in his browser. He reported that a response was never returned. Troubleshooting the issue, we determined that internally to the API call, the execution of `zmu` was seg faulting. He demonstrated this by running `zmu` from the command line on his Zoneminder server. As you can see, it returned the alarm status, then segfaulted.
```
zmu -s -m8
4
05/14/20 13:24:59.519175 undef[26596].ERR-zm_signal.cpp/52 [Got signal 11 (Segmentation fault), crashing]
```

When invoked through the API, this failure causes the API call to hang, and results in a orphaned database connection. Over time, after numerous calls to the API, Zoneminder will run out of database connections.

I realize this is an issue with the user's Zoneminder installation. Something in his setup obviously is corrupted, and he's in the process of trying to fix his installation. I'm logging this issue to bring some visibility to the problem. It would be highly desirable that a failure such as this does not cause the API call to hang, nor should it orphan a database connection.

I also completely understand if you want to close this issue with the argument that a corrupted installation can result in unpredictable behavior.

connortechnology, 2020-05-14:
First off, thanks for the detailed excellent issue report.  

ZMU should not crash without at least giving a good explanation why.  

We might need to turn on debug logging at a level of 3 or more to get a better idea of where in the code it is crashing. 

Also need to understand why a zmu segfault would then hang the api... 

I'll take  a look and see what I can figure out.

mhilbush, 2020-05-14:
Thanks. Hopefully, my user hasn't fixed the issue yet. I just asked him to hold off on fixing it so that we can capture some additional debug logs. I should've thought to do that. Sorry.

connortechnology, 2020-05-14:
confirmed.  I can reproduce this here.

connortechnology, 2020-05-14:
Have a fix.  Releasing 1.34.13.

mhilbush, 2020-05-14:
Thanks for the quick turnaround in reproducing and fixing the issue! I'm amazed you were able to track it down so quickly.

Perhaps you can help me understand a little better why the issue was occurring in some cases but not in others. For example, it happens consistently for one user of my app, but I've never seen it occur (nor has it been reported by any other users of my app).

I did ask the user to hold off on trying to workaround the issue, so that he can verify the fix when 1.34.13 comes out.

mhilbush, 2020-05-14:
I just installed 1.34.13. I'm seeing something that doesn't look right.

When invoking this in a browser window.
```
http://zoneminder/zm/api/monitors/alarm/id:1/command:status.json
```
I get this response
```
{"status":"Exitingclosing db"}
```

Similarly, running `zmu -s -m1` from the shell results in this output.
```
mark@server2:~$ sudo zmu -s -m1
0
Exitingclosing db
```

erikkastelec, 2020-05-15:
I also encountered Exitingclosing db response when upgrading from 1.34.12 to 1.34.13, which brakes Homeassistant integration.

connortechnology, 2020-05-15:
Please try out 1.34.14

connortechnology, 2020-05-15:
I can't shed too much light on it.  The issue is in the mysql client libs.  Calling mysql_library_end(); shouldn't segfault and hang, but apparently it does.  Lots of other reports about it, and mysql's position seems to be to not call it.  Just let it leak ram.  So be it.

mawa85, 2020-05-15:
can't find it in the focal repo

erikkastelec, 2020-05-15:
1.34.14 fixed it for me. Thank you.

mhilbush, 2020-05-15:
One of my users confirms the DB connection issue is resolved in 1.34.14. Oops, I see he already commented above. 

> mysql's position seems to be to not call it

@connortechnology Ah it's one of those *"doctor, doctor, it hurts when I do that"* things... 

In any event thanks for the quick fix!

closed 2020-05-15 by mhilbush

bhearsum, 2020-05-30:
Thank you thank you!

Home-Control9000, 2021-12-10:
Im running v1.37.5 and having exactly the same problem.  
im also using: http://zoneminder/zm/api/monitors/alarm/id:x/command:status.json
I got many hanging zmu processes in 1 day and it wil take 1 sql connection. With a max of 150 the system is running out of available connections and i have to restart apache.