ZoneMinder/zoneminder #2940: Web console crash after Montage due to web browser connection limit / Systems with more than 6 cameras

bartbk, 2020-05-23:
**Describe Your Environment**
Version of ZoneMinder [1.34.15]
How you installed ZoneMinder [e.g. PPA iconnor]
Full name and version of OS Ubuntu 20.04
Browser name and version (Chrome Version 81.0.4044.138 (Official Build) (64-bit) / Edge Version 84.0.516.0 (Official build) canary (64-bit), Internet Explorer 11.836.18362.0 )

**If the issue concerns a camera**
Its camera independent.

**Describe the bug**
Web console hangs completely and becomes unresponsive in browser that have limit of 6 connections per server / domain when you have more than 6 cameras defined, when you click Montage. In browsers with higher connection limit (Internet Explorer) it works normally - propably up to number of cameras bigger of connection limit of that browser.

**To Reproduce**
Steps to reproduce the behavior:
1. Define 7 or more cameras with streams - could be file if you do not have enough cameras. All should be enabled. 
2. Open ZM Web console with browser with 6 connection limit (Chrome or Edge)
3. Click on Montage tab (Not Montage review - in Montage review it works)
4. Browser becomes unresponsive - no error in log
5. Open this same ZM Web console in IE11 it should work as designed - IE11 opens 7 streams in Montage, and Web console stays responsive.

**Expected behavior**
Responsive browser in Montage at systems with more than 6 cameras.
As the browser connection limit is rather not easy to change the expected behavior should be hard limitation to 6 concurrent streams in Montage or other solution proposed. 

**Debug Logs**
```
There is no logs until browser is open.

After closing problematic page in a browser we can observe error below for many PIDs, probably from orphaned processes after web console crash.

Date/Time | Component | Server | PID | Level | Message | File | Line
-- | -- | -- | -- | -- | -- | -- | --
2020-05-23 06:48:23 | web_php |   | 928812 | ERR | Socket /var/run/zm/zms-773833s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. | /usr/share/zoneminder/www/includes/functions.php | 2177

```

connortechnology, 2020-05-23:
Known problem with a documented solution.
https://zoneminder.readthedocs.io/en/stable/userguide/options/options_network.html

closed 2020-05-23 by connortechnology

bartbk, 2020-05-24:
Documented solution is how to allow over 6 connection. But don't you think that web console should not crash without that and become completely unresponsive?

connortechnology, 2020-05-24:
It's not something I can fix.  Take it up with chrome.  This issue isn't new, and has been discussed ad nauseum.