ZoneMinder/zoneminder #2941: Console view is broken on small devices

roddone, 2020-05-26:
**Describe Your Environment**
- Version of ZoneMinder : 1.34.12
- installed from PPA iconnor
- Ubuntu Server 20.04
- Chrome 83.0.4103.61 (64 bits) & Firefox 76.0.1 (64 bits) & Microsoft Edge 83.0.478.37

**Describe the bug**
On small devices (like mobile phones), the console view's responsive display is broken because the grid that contains the list of monitors is larger than the screen. It makes it possible to scroll right/left the entire page .
I principaly use ZM on my mobile phone and this glitch is really annoying 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the console view with a mobile phone (or simulate one with your browser's dev tool)
2. Add a monitor if you don't already have one
3. Scroll left/right
4. See error
[image]

**Expected behavior**
In a perfect world, the whole interface should be done "mobile first", but we could also simply wrap the table in a div with class "table-responsive" (see https://getbootstrap.com/docs/3.3/css/#tables-responsive )
This would make the table scroll instead of the whole page.

roddone, 2020-05-26:
There are also a lot of other "glitches" or possible visual improvements when using ZM on tablet / mobile.
Would it be relevant to describe each one in a separate issue or group them in a single one ?

connortechnology, 2020-05-26:
I think technically each should have it's own issue, although if some are similar feel free to group them a bit.

Are you intending to solve these issues yourself?

roddone, 2020-05-27:
Of course I will try to fix these issues but i only use a small bench of ZM's views like console and watch views, so I won't be able to review all the views.

I will open issues when I identify problems ;)

PR #2942 (2020-06-03): make monitors table (console view) responsive
wrap the monitors table in a `<div class="table-responsive">...</div>` to ensure the tables scrolls instead of the whole page

closes #2941
closed 2020-06-03 by connortechnology