ZoneMinder/zoneminder #2951: Incorrect use of quotations in zmupdate.pl causes SQL dump to fail on MariaDB.

brianara3, 2020-06-02:
**Describe Your Environment**
- ZM version 1.34.14
- Yum install from RPMFusion
- CentOS 7.7
- MariaDB 5.5.65

**Describe the bug**
After upgrading from NM 1.34.5 to 1.34.14 I am directed to run the zmupdate.pl script. This fails on the mysqldump command due to the fact the script inserts double quote marks (") around the password, and mysqldump interprets them as part of the password. This causes an access denied error while trying to complete the dump. If I run the command manually and swap the double quotes for single ones the command runs fine. I don't know enough about Perl to fix the proper quote syntax.

**To Reproduce**
Steps to reproduce the behavior:
1. Run zmupdate.pl
2. Answer yes to the SQL backup
3. Script fails if mysql not looking for double quotes in password.

**Expected behavior**
Script shouldn't fail to run backup?

**Debug Logs**
```
Initiating database upgrade to version 1.34.14 from version 1.34.5

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop :

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /var/lib/zoneminder/temp if you have a lot of events.
Press 'y' for a backup or 'n' to continue : y
Creating backup to /var/lib/zoneminder/temp/zm-1.34.5.dump. This may take several minutes.
mysqldump: Got error: 1045: "Access denied for user '<**db user**>'@'localhost' (using password: YES)" when trying to connect
Output:
Command 'mysqldump -hlocalhost -u<**db user**> -p"<**db pass**>" --add-drop-table --databases zm > /var/lib/zoneminder/temp/zm-1.34.5.dump' exited with status: 2

```

brianara3, 2020-06-02:
Looks like the script also fails to run the actual upgrade due to the same reason.

-p"<password>"

connortechnology, 2020-06-03:
it seems to have the password wrong.  Assuming that the password used was actually correct... does it have odd characters in it?

connortechnology, 2020-06-03:
I think the $ is causing the shell to treat it as a variable.  So I'll have to see what I can do to escape it.

connortechnology, 2020-06-03:
Check current master.  We now use single quotes, and escape any single quotes in the password.

brianara3, 2020-06-04:
Looks like it would work... but I can't run it again until next DB upgrade :) Just gives me an error that I'm already at current version.

connortechnology, 2020-06-04:
Should be a new release in the next day or two.

knight-of-ni, 2020-06-06:
zm 1.34.16 has been pushed to rpmfusion testing. Please test!

connortechnology, 2020-06-11:
Please test with 1.34.16 and close this if your issues are resolved. I believe they will be.

brianara3, 2020-06-12:
Looks like that resolved the issue. Thank you.

closed 2020-06-12 by brianara3