ZoneMinder/zoneminder #2960: apt upgrade zoneminder fails

jeroenst, 2020-06-12:
**Describe Your Environment**
- Version of ZoneMinder [1.3.15-bionic1]
- How you installed ZoneMinder [PPA]
- Full name and version of OS [Linux mint 19.3]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Run apt update
2. Run apt upgrade zonemider
3. See error

**Expected behavior**
Zoneminder restarts after upgrade

**Debug Logs**
```

root@server:~# apt upgrade zoneminder
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zoneminder is already the newest version (1.34.16-bionic1).
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up zoneminder (1.34.16-bionic1) ...
Detected db service is mysql.service
mysqladmin: connect to server at 'localhost' failed
error: 'Access denied for user 'root'@'localhost' (using password: NO)'
Creating zm db
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: NO)
Error creating db.
dpkg: error processing package zoneminder (--configure):
 installed zoneminder package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 zoneminder
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

jeroenst, 2020-06-12:
I worked around this problem:
1. Stop mysqld
2. Starting mysql with "mysqld_safe --skip-grant-tables &"
3. Run apt-get upgrade
4. Kill mysqld_safe
5. Start mysqld
6. run zmupdate.pl
7. Start zoneminder

closed 2020-06-15 by connortechnology

jeroenst, 2020-06-15:
Closed? The issue still persists....

connortechnology, 2020-06-15:
You have removed the system maintenance account from mysql.  Not much we can do about that.  If you lock down mysql, then it is up to you to create the db.

jeroenst, 2020-06-15:
Ok, thank you.

Installing Shinobi caused this, it removed mysql and installed Mariadb which I didn't want. 

After I removed Shinobi an reinstalled mysql this error occurred. 

Reinstalling mysql completely with keeping databases hopefully has solved this problem.