ZoneMinder/zoneminder #2961: Coordinates are covering motion settings in zone configuration

jeroenst, 2020-06-12:
**Describe Your Environment**
- Version of ZoneMinder [1.34.16]
- How you installed ZoneMinder [PPA]
- Full name and version of OS [Linux mint 19.3]
- Browser name and version [Chrome Version 83.0.4103.61 (Official Build) (64-bit)]

**Describe the bug**
When I open a zone, the coordinates of the zones covers the settings of the motion detection

**To Reproduce**
Steps to reproduce the behavior:
1. Go to zone
2. Open Zone
3. Scroll down
4. See fault

**Expected behavior**
The coordinates shouldn't cover the motion settings.

jeroenst, 2020-06-12:
file:///home/jeroen/Pictures/Screenshot%20from%202020-06-12%2014-03-06.png

jeroenst, 2020-06-12:
[image]

connortechnology, 2020-06-15:
Can you right click inspect and tell us the width of the settsPanel div and the zonePoints div?
They should each be 600px.
The layout here is pretty simple... 600px wide, floated left.  They should not overlap. If there were going to, the zonePoints should drop down below.

jeroenst, 2020-06-23:
[image]
[image]

jeroenst, 2020-06-23:
The issue occurs in Chrome, there is no problem in Firefox.

connortechnology, 2020-08-06:
I now understand what is going on here.  A quick fix is in master, will be merged to 1.34.19.

jeroenst, 2020-08-06:
Thanx!

closed 2020-08-25 by connortechnology

jeroenst, 2020-11-30:
I can confirm this bug is solved in v 1.34.20.

Thanx!