ZoneMinder/zoneminder #2963: HTTP Error 500 after update

connectnet, 2020-06-17:
Version 1.35.5 
Calling Filter makes an Error 500 after update.

connortechnology, 2020-06-17:
Please consult your apache logs.

connectnet, 2020-06-17:
'[Wed Jun 17 18:38:44.974191 2020] [php7:warn] [pid 29842] [client 127.0.0.1:50978] PHP Warning:  require_once(includes/User.php): failed to open stream: No such file or directory in /usr/share/zoneminder/www/skins/classic/views/filter.php on line 30, referer: http://127.0.0.1/zm/index.php?view=console
[Wed Jun 17 18:38:44.974287 2020] [php7:error] [pid 29842] [client 127.0.0.1:50978] PHP Fatal error:  require_once(): Failed opening required 'includes/User.php' (include_path='.:/usr/share/php') in /usr/share/zoneminder/www/skins/classic/views/filter.php on line 30, referer: http://127.0.0.1/zm/index.php?view=console'

connortechnology, 2020-06-17:
Forgot to add a file that I had created.  Current master now has it.  packages are building.

closed 2020-06-17 by connectnet