ZoneMinder/zoneminder #2965: Filters web link no longer working.  Missing Includes/User.php file

KTownKnerd, 2020-06-17:
**Describe Your Environment**
-  Zoneminder Version:1.35.5~20200616.14-buster
- Debian Repository (deb https://zmrepo.zoneminder.com/debian/master buster/)
- Debian 10 Buster
**Describe the bug**
Filters interface from the web interface is no longer displayed.  When clicking Filters I receive the following error: This page isn’t working dvr is currently unable to handle this request.
HTTP ERROR 500.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'web interface'
2. Click on 'Filters'
3. See error

**Expected behavior**
Clicking the Filters link should display the Filters interface.

**Debug Logs**
```
2020/06/17 19:02:07 [error] 588#588: *17365 FastCGI sent in stderr: "PHP message: PHP Warning:  require_once(includes/User.php): failed to open stream: No such file or directory in /usr/share/zoneminder/www/skins/classic/views/filter.php on line 30PHP message: PHP Fatal error:  require_once(): Failed opening required 'includes/User.php' (include_path='.:/usr/share/php') in /usr/share/zoneminder/www/skins/classic/views/filter.php on line 30" while reading response header from upstream, client: 192.168.219.209, server: _, request: "GET /zm/index.php?view=filter&sort_field=StartTime&sort_asc=1 HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.3-fpm.sock:", host: "dvr", referrer: "http://dvr/zm/index.php"

```

connortechnology, 2020-06-18:
You are not up to date. Please update.  I think you will find that it has been added.

KTownKnerd, 2020-06-18:
Hello,
I can confirm that I upgraded to 1.35.5~20200617.19-buster and it is now working. 
Thank you.

closed 2020-07-07 by connortechnology