ZoneMinder/zoneminder #2967: Question Regarding Extra Large Timestamp Font Size

p0xus, 2020-06-19:
I apologize if this is the improper way to ask this, but I have been doing some research on how to increase the timestamp font size from "Large" in order to make it more readable with 1080p cameras, playing back at reduced resolutions. 
In this vein of research, I have found Issue #992 in which schrorg modified the regular font files and created the large font. My question is whether there is any reason for me to NOT replicate his work and create an extra-large font type.
Other then the time-consuming work of recreating the font characters with a larger size.

Thanks for all your work in maintaining and improving ZM.

connortechnology, 2020-06-19:
I was thinking about investigating whether we could use some font drawing library that would allow us to use variable sizes and anti-aliasing etc.... would have to be performant though. I have no idea when I would get around to that though.

So I would say that either you create the extra large font, or perhaps look into code that would scale up an existing bitmap, or do that font drawing library investigation.

p0xus, 2020-06-19:
I'll look into either scaling up a bitmap or a font drawing library; but my coding capability is rather limited, so I think it most likely that I'll end up replicating schrorg's work. I agree, however, that it would be best to have a more variable system in place, but I think my skills are too limited for that. 

Thanks for your advice.

connortechnology, 2020-12-25:
We have a larger font size now.

closed 2020-12-25 by connortechnology