ZoneMinder/zoneminder #2968: crash with res at 1080x1920

connortechnology, 2020-06-21:
See forums for original reporter. 
https://forums.zoneminder.com/viewtopic.php?f=40&t=29613
1088x1920 and 1072x1920 do not crash.  

Isaac does not have any cameras that will do actual rotation in camera.

neillbell, 2020-07-15:
I have a HikVision DS-2CD2442FWD-IW that does allow in-camera rotation. I just configured it for 1080x1920 in the camera and also in ZoneMinder and it worked fine. Is this configuration what is expected to crash?

connortechnology, 2020-07-16:
I believe there is a further requirement that you need to be converting to 24bit RGB instead of 32bit RGBA.  

In any event, the fix is already in master, just undergoing some testing while I finish everything else that is going in 1.34.17

mastertheknife, 2020-07-18:
Hi, i doubt its the limitation, because 1920*1080 = 2073600 and its dividable by 12, 16 and also 64.

connortechnology, 2020-07-18:
@mastertheknife nice to see you.

I believe I already have the problem fixed in master.

The issue is that the WIDTH needs to be aligned to use sws_scale.  Not width*height.  sws_scale acts line by line.

connortechnology, 2020-07-24:
Ugh.  I have come to believe that my fix just won't work.  Going to have to do a memory copy instead.

connortechnology, 2020-07-24:
Okay, current status which I think I can port to 1.34.17 is that we will now leave the shared images as a contiguous block of ram and do a memcpy in zm_ffmpeg_camera if required.

closed 2020-08-11 by connortechnology

tqk2811, 2022-06-14:
how caculate the width? my sws_cale crash too