ZoneMinder/zoneminder #2970: Link between nodect and modect monitors breaks

sabretoothedhamster, 2020-06-22:
**Describe Your Environment**
ZoneMinder 1.34.15 PPA
Ubuntu 18.04.4 LTS 4.15.0-106 x86_64

**Describe the bug**
Two monitors attached to the same camera and configured in modect/nodect link break apart if modect monitor is stuck and restarted by zmwatch. After modect monitor is restarted nodect one does not "see" it's alarms anymore.

**To Reproduce**
Create modect and notect monitors to the same camera. Make link from nodect to modect monitor. Simulate somehow modect monitor troubles (which are diagnosed by zmwatch as "Restarting capture daemon for Cam1, time since last capture 11 seconds" but modect process is live). Then this monitor will be restarted by zmwatch, but linked nodect monitor will not, and will loose possibility to capture alarms generated by it's modect counterparty.

**Expected behavior**
After restart link keeps working.

**Debug Logs**
In the attached file (partial zmwatch debug log) Cam*d are modect monitors for respective Cam*s nodect monitors. After some network glitch several monitors were restarted but only Cam1d, Cam6d and Cam3d lost their connectivity to Cam1s, Cam6s and Cam3s respectively after 15:14:07

I do have full debug logs for Cam1 modect and nodect monitors (both zma and zmc) and ready to provide them in full or partially if needed.

```

sabretoothedhamster, 2020-06-22:
zmwatch debug log part

[zm_debug.log.906_part.log](https://github.com/ZoneMinder/zoneminder/files/4810637/zm_debug.log.906_part.log)

connortechnology, 2020-06-22:
Good find.  The fix is fairly trivial.  I'll have it fixed soon.

sabretoothedhamster, 2020-06-23:
Thanks a lot Isaac! Would you please post a note here with a link to respective commit, so I will be able to apply it before next full build?

connortechnology, 2020-07-07:
The commit is 31a3be5c8eba05a6dd02df54a2837c2d85835872.   It will be included in 1.34.17 which should be released soon.
In the meantime you could try master.

sabretoothedhamster, 2020-07-15:
I may give it a try but do not want to go with complete package uninstall/install. Is it safe to put master binaries (zmc, zma, zmu, zms) over the existing ones from v1.34.16? If not then I will wait for 1.34.17 since I have implemented myself some quick fix (in zmwatch.pl to unconditionally restart linked monitors) and currently do not suffer from this bug...

connortechnology, 2020-07-15:
No you can't just drop in master binaries.  Of course 1.34.17 will overwrite your zmwatch as well.

connortechnology, 2020-08-06:
1.34.17 has been released and should fix this issue.  Please verify and close if so.

closed 2021-03-10 by connortechnology