ZoneMinder/zoneminder #2972: Axis API v2 Password encryption

tobelger, 2020-07-02:
**Describe Your Environment**
- Zoneminder 1.34.16 
- PPA install
- Ubuntu 20.04 Server
- Firefox ESR, Chrome

**If the issue concerns a camera**
- Axis (all but here in special M2026-LE Mk II)
- 30 FPS
- 1920 x1080

**Describe the bug**
Axis PTZ function is supported by Zoneminder via AXIS API V2 Driver. If i add a newer Model of Axis with activated PTZ to my Enviroment, i get a PTZ ERROR 401 Unauthorized. Credentials are right.
The Problem is also described here: https://www.marmosetelectronics.com/computing/axis-m5525-e-ptz-vs-zoneminder/
The explained Workaround worked for me, but u have du disable every Passwordencryption. The Mode "encrypted+unencrypted " doesnt worked for me.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console'
2. Click on 'Monitorsettings'
3. go to PTZ-Tab and configure your AXIS-Camera PTZ
4. Try to move Camera
5. See Error in Logs

**Expected behavior**
The Axis API V2 Driver need to be updated to work with Digest.

**Debug Logs**
```
2020-07-02 16:35:53 | zmcontrol_11 |   | 153292 | ERR | Error cmd http://*****:*****@192.168.*.*:80/axis-cgi/com/ptz.cgi?rtilt=40 failed: '401 Unauthorized' | zmcontrol.pl
-- | -- | -- | -- | -- | -- | --
```

connortechnology, 2020-07-07:
The Axis V2 script was updated to support Digest.  

Is you password perhaps complex with weird characters?  Please try with a simple one.

Would we be able to get access to your system to try to fix it?

tobelger, 2020-07-07:
Yes i can give you an Account and Link. I will send you an email with Credentials to login.

connortechnology, 2020-07-08:
Will be included in 1.34.17

closed 2020-07-08 by connortechnology

tobelger, 2021-06-18:
The Error ist back with 1.34.26

connortechnology, 2021-06-18:
I don't think there are any relevant commits between 1.34.17 and 1.34.26.  We are going to have to start from scratch in diagnosing.

tobelger, 2021-06-18:
I added the two files. The AxisV2.pm.13426.txt
[AxisV2.pm.13426.txt](https://github.com/ZoneMinder/zoneminder/files/6678324/AxisV2.pm.13426.txt)
is the File which came with update to 1.34.26. The AxisV2.pm.txt 
[AxisV2.pm.txt](https://github.com/ZoneMinder/zoneminder/files/6678327/AxisV2.pm.txt)
is the restored backup you created last Year. The AxisV2.txt as pm-File works fine without any Issues. It is the Githubversion from July 08 2020.

commit e5afc1a2 (2020-07-08): Cleanup code that sets digest credentials.  Fixes #2972