ZoneMinder/zoneminder #2973: 'zmc -d /dev/video0' crashed, signal 132 -- on 32-bit ASUS Eee PC

ksvu, 2020-07-05:
**Environment**
ASUS Eee PC 2G Surf
ZoneMinder v1.34.16 from ppa
Lubuntu 32-bit 18.04.4 LTS
```
# uname -srvmpio
Linux 4.15.0-108-generic #109-Ubuntu SMP Fri Jun 19 11:36:05 UTC 2020 i686 i686 i686 GNU/Linux
```
Memory (ZM works)
```
# free -m
              total        used        free      shared  buff/cache   available
Mem:            418         194          13          12         209         199
Swap:          1765         154        1611
```
I do not use internal EeePC drive, but instead boot from external USB hdd with 2 partitions one for the system and the rest for ZM 
```
# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0   1.9G  0 disk 
sdb      8:16   0 298.1G  0 disk 
├─sdb1   8:17   0  37.3G  0 part /
└─sdb2   8:18   0 260.9G  0 part /mnt/dvr
```
I operate ZM from another machine in the same LAN with
Mozilla Firefox 75.0 (64-bit) for Ubuntu canonical - 1.0

**camera**
works with Cheese without issues
```
# v4l2-ctl -d /dev/video0 --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : USB2.0 PC CAMERA: USB2.0 PC CAM
	Bus info      : usb-0000:00:1d.7-2.3.1
	Driver version: 4.15.18
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=130 value=130
[... 80 lines cut ...]
2020-07-04 19:55:56.047870 zmdc[1650].INF-zmdc.pl ['zmc -d /dev/video0' started at 20/07/04 19:55:56]
2020-07-04 19:55:56.042280 zmdc[1622].INF-zmdc.pl ['zmc -d /dev/video0' starting at 20/07/04 19:55:56, pid = 1650]
2020-07-04 19:55:53.970130 zmpkg[1608].INF-zmpkg.pl [Single server configuration detected. Starting up services.]
2020-07-04 19:55:53.780280 zmdc[1622].INF-zmdc.pl [Socket should be open at /var/run/zm/zmdc.sock]
2020-07-04 19:55:52.104830 zmdc[1622].INF-zmdc.pl [Server starting at 20/07/04 19:55:52]
2020-07-04 19:55:47.739350 zmpkg[1608].INF-zmpkg.pl [Command: start]
```
And here is final part of /var/log/zm/zmc_dvideo0.log The timestamps do not correspond to previous, but it goes in cycles... 
```
07/04/20 22:36:30.921712 zmc_dvideo0[6885].DB4-zm_image.cpp/335 [Delta: Using SSSE3 delta functions]
07/04/20 22:36:35.129450 zmc_dvideo0[6887].DB1-zm_logger.cpp/255 [LogOpts: level=DB9 effective=DB9, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zmc_dvideo0.log, syslog=DB9]
07/04/20 22:36:35.134090 zmc_dvideo0[6887].DB1-zm_utils.cpp/282 [Detected a x86\x86-64 processor with AVX2]
07/04/20 22:36:35.134548 zmc_dvideo0[6887].DB1-zm_monitor.cpp/1992 [Loading Monitors with SELECT `Id`, `Name`, `ServerId`, `StorageId`, `Type`, `Function`+0, `Enabled`, `LinkedMonitors`, `AnalysisFPSLimit`, `AnalysisUpdateDelay`, `MaxFPS`, `AlarmMaxFPS`,`Device`, `Channel`, `Format`, `V4LMultiBuffer`, `V4LCapturesPerFrame`, `Protocol`, `Method`, `Options`, `User`, `Pass`, `Host`, `Port`, `Path`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`+0, `Deinterlacing`, `DecoderHWAccelName`, `DecoderHWAccelDevice`, `RTSPDescribe`, `SaveJPEGs`, `VideoWriter`, `EncoderParameters`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`,`ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `MinSectionLength`, `FrameSkip`, `MotionFrameSkip`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `TrackMotion`, `Exif`, `SignalCheckPoints`, `SignalCheckColour` FROM `Monitors` WHERE `Function` != 'None' AND `Type` = 'Local' AND `Device`='/dev/video0']
07/04/20 22:36:35.140012 zmc_dvideo0[6887].DB4-zm_db.cpp/112 [Success running query: SELECT `Id`, `Name`, `ServerId`, `StorageId`, `Type`, `Function`+0, `Enabled`, `LinkedMonitors`, `AnalysisFPSLimit`, `AnalysisUpdateDelay`, `MaxFPS`, `AlarmMaxFPS`,`Device`, `Channel`, `Format`, `V4LMultiBuffer`, `V4LCapturesPerFrame`, `Protocol`, `Method`, `Options`, `User`, `Pass`, `Host`, `Port`, `Path`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`+0, `Deinterlacing`, `DecoderHWAccelName`, `DecoderHWAccelDevice`, `RTSPDescribe`, `SaveJPEGs`, `VideoWriter`, `EncoderParameters`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`,`ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `MinSectionLength`, `FrameSkip`, `MotionFrameSkip`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `TrackMotion`, `Exif`, `SignalCheckPoints`, `SignalCheckColour` FROM `Monitors` WHERE `Function` != 'None' AND `Type` = 'Local' AND `Device`='/dev/video0']
07/04/20 22:36:35.140692 zmc_dvideo0[6887].DB1-zm_monitor.cpp/2000 [Got 1 monitors]
07/04/20 22:36:35.141164 zmc_dvideo0[6887].DB1-zm_monitor.cpp/2116 [Got 1 for v4l_captures_per_frame]
07/04/20 22:36:35.141643 zmc_dvideo0[6887].DB2-zm_camera.cpp/55 [New camera id: 1 width: 640 height: 480 colours: 3 subpixelorder: 6 capture: 1]
07/04/20 22:36:35.142136 zmc_dvideo0[6887].DB2-zm_local_camera.cpp/346 [V4L support enabled, using V4L2 api]
07/04/20 22:36:35.142529 zmc_dvideo0[6887].DB2-zm_local_camera.cpp/365 [little-endian processor detected]
07/04/20 22:36:35.142901 zmc_dvideo0[6887].INF-zm_local_camera.cpp/447 [No direct match for the selected palette (0x00000001) and target colorspace (03). Format conversion is required, performance penalty expected]
07/04/20 22:36:35.171515 zmc_dvideo0[6887].DB2-zm_local_camera.cpp/455 [Using swscale for image conversion]
07/04/20 22:36:35.172152 zmc_dvideo0[6887].DB3-zm_local_camera.cpp/654 [Selected subpixelorder: 6]
07/04/20 22:36:35.178203 zmc_dvideo0[6887].DB4-zm_image.cpp/273 [Blend: Using SSE2 fast blend function]
07/04/20 22:36:35.178904 zmc_dvideo0[6887].DB4-zm_image.cpp/335 [Delta: Using SSSE3 delta functions]
```

connortechnology, 2020-07-05:
signal 132 is illegal instruction.  I suspect that your processor doesn't support the SSE3 instructions that we are trying to use. @mastertheknife 

Can we get a cat /proc/cpuinfo?

ksvu, 2020-07-05:
Sure, here it is
```
# cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 13
model name	: Intel(R) Celeron(R) M processor          800MHz
stepping	: 8
microcode	: 0x20
cpu MHz		: 570.034
physical id	: 0
siblings	: 1
core id		: 0
cpu cores	: 1
apicid		: 0
initial apicid	: 0
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 2
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov clflush dts acpi mmx fxsr sse sse2 ss tm pbe nx bts cpuid pti
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 1140.06
clflush size	: 64
cache_alignment	: 64
address sizes	: 32 bits physical, 32 bits virtual
power management:

```

ksvu, 2020-07-05:
Thank you for the hint by the way :-)
I disabled CPU_EXTENSIONS in (ZM - Options)->Config and got the Monitor-1 working.
Is this the best solution?

connortechnology, 2020-07-05:
It's the best short term solution.  The best solution is for us to fix our sse detection.

ksvu, 2020-07-05:
Good. So I feel like closing this issue, right?

connortechnology, 2020-07-05:
No please don't.  Apparently I can get access to a 701 model to test with.  I believe our cpu detection code is crap.

connortechnology, 2020-07-05:
On this hardware we probably need all the hardware support we can get.  Turning off optimisations... is going to limit the usefulness.

SteveGilvarry, 2020-07-05:
Maybe make a something to test this? https://gist.github.com/hi2p-perim/7855506

connortechnology, 2020-08-06:
I have merged a fix to master for this.  It didn't make it in for 1.34.17 or .18.  Maybe it will for .19.

closed 2022-04-17 by stale[bot]