ZoneMinder/zoneminder #2978: Alarm via API returns wrong value

tranzmatt, 2020-07-13:
**Describe Your Environment**
- Version of ZoneMinder 1.34.16
- How you installed ZoneMinder  PPA APT install
- Full name and version of OS  NVidia Jetpack 32.3.1 (running Bionic 18.04)

**Describe the bug**

Call made via pyzm:

make_request called with url=http://192.168.1.247/zm/api/monitors/alarm/id:3/command:on.json payload={} type=get query={}

Where mid=3. I get a return json of something like

ARM: {'status': 'Alarmed event id: 2'}

But in the web-ui, ID #3 seems to have been created instead.

**Expected behavior**
Shouldn't "3" be returned in the json string?

pliablepixels, 2020-07-13:
To be specific, this is not an API issue. `zmu` returns the wrong alarm eid when used to trigger an alarm (which then filters up via the API layer too)

tranzmatt, 2020-07-13:
Ok.  Should I close this and open something different or has it already been reported?

pliablepixels, 2020-07-13:
No that’s fine. Let’s keep it as is.

commit 3658c3b5 (2020-07-16): Update state to ALARM AFTER we have opened the event, so that we get the correct event Id.  Fixes #2978 Fixes #2894
commit d849ee54 (2020-07-16): Update state to ALARM AFTER we have opened the event, so that we get the correct event Id.  Fixes #2978 Fixes #2894
closed 2020-07-16 by connortechnology