ZoneMinder/zoneminder #2982: Users API endpoints return 500 "Call to a member function allow() on null"

matthewtgilbride, 2020-07-19:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.34.16
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
Via the installation script I wrote here: https://github.com/matthewtgilbride/zoneminder-aws/blob/master/zminstall.sh
- Full name and version of OS
Ubuntu 18.04
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
See description:
```
{
    "success": false,
    "data": {
        "name": "An Internal Error Has Occurred.",
        "message": "An Internal Error Has Occurred.",
        "url": "/zm/api/users.json?token=<my-token>",
        "exception": {
            "class": "Error",
            "code": 0,
            "message": "Call to a member function allow() on null"
        }
    }
}
```

**To Reproduce**
```
curl --location --request GET 'http://<host>/zm/api/users.json?token=<token>
```

**Expected behavior**
This seems like a pretty simple issue, but I don't know PHP.

pasichnykvasyl, 2020-07-22:
I have this problem too

connortechnology, 2020-08-17:
I think this might have gotten into 1.34.19.  Can you test and comment?

connortechnology, 2020-08-17:
Wait nope, git history is really confused.  This is scheduled for 1.34.20.  Has been pushed to master.

commit bf163bee (2020-08-17): Update permissions checking when viewing/editing users.  Fixes #2982.  Also add a beforeFind that will remove the Password field from the query so that we don't return passwords
commit 4c69d7b7 (2020-08-18): Merge pull request #3001 from connortechnology/update_users_controller

Update permissions checking when viewing/editing users.  Fixes #2982.…
closed 2020-08-18 by connortechnology