ZoneMinder/zoneminder #2983: Zone create API endpoint returns 200, but no zone is created

matthewtgilbride, 2020-07-20:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.34.16
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
Via the installation script I wrote here: https://github.com/matthewtgilbride/zoneminder-aws/blob/master/zminstall.sh
- Full name and version of OS
Ubuntu 18.04
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
I can POST to the API endpoint to create a zone and get a 200 back.  However, when I subsequently look in the console there are no zones for the monitor

**To Reproduce**
Example curl:
```
curl --location --request POST 'http://ec2-3-236-233-51.compute-1.amazonaws.com/zm/api/zones.json?token=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJab25lTWluZGVyIiwiaWF0IjoxNTk1MjU5NjUwLCJleHAiOjE1OTUyNjY4NTAsInVzZXIiOiJhZG1pbiIsInR5cGUiOiJhY2Nlc3MifQ.37ITrWmtYc070O1XPw4n5em_WpLLgtKUGBNcDUJ9DY4' \
--header 'Content-Type: application/x-www-form-urlencoded' \
--data-raw 'MonitorId=1&Zone%5BName%5D=Foo&Zone%5BType%5D=Active&Zone%5BUnits%5D=Percent&Zone%5BNumCoords%5D=4&Zone%5BCoords%5D=512%2C537%201917%2C575%201919%2C1079%200%2C1077&Zone%5BArea%5D=872664&Zone%5BAlarmRGB%5D=16711680&Zone%5BCheckMethod%5D=Blobs&Zone%5BMinPixelThreshold%5D=20&Zone%5BMinAlarmPixels%5D=69813&Zone%5BFilterX%5D=3&Zone%5BFilterY%5D=3&Zone%5BMinFilterPixels%5D=52359&Zone%5BMinBlobPixels%5D=43633&Zone%5BMinBlobs%5D=1&Zone%5BOverloadFrames%5D=0&Zone%5BExtendAlarmFrames%5D=0'
```

**Expected behavior**
After this endpoint returns 200 I would expect to see a new zone attached to the monitor in the console, but do not.

matthewtgilbride, 2020-07-20:
Interesting - when I use a body from the example in the docs, it works.  When I use the body specified above (with the correct monitorId), it seems to fail silently.

matthewtgilbride, 2020-07-20:
Nevermind...(or perhaps not...feel free to close if you see fit).

This is caused by incorrectly specifying the body parameter for the monitorId (I had MonitorId instead of Zone[MonitorId]).  I would expect a 400 or 404 in that case.

connortechnology, 2020-07-20:
Please note that the --location argument will NOT do a post on the redirect.  So I don't suggest using that.
PR incoming to add validation.

commit 955fc00b (2020-07-20): Fix Zone add api.  Restart zmc when adding a zone.  Use validation. Fixes #2983
closed 2020-07-20 by connortechnology