ZoneMinder/zoneminder #2984: H264 passthrough event metadata at end of video file

okgo88, 2020-07-21:
----------------------------------------------------------------------------------------------------

**Describe Your Environment**
- 1.34.16
- PPA
- Ubuntu 18.04.4 LTS
- Chrome Version 83.0.4103.116 (Official Build) (64-bit)

**If the issue concerns a camera**
- N/A
- N/A
- N/A
- H264 passthrough

**Describe the bug**
In previous versions of Zoneminder, playing events from "H264 passthrough" recorded cameras would start immediately, allow scrubbing, and basically stream the requested segments in a very pleasant way. In more current versions of Zoneminder the videos need to download completely (minutes to hours depending on the connection) before they will be viewable. I've narrowed down the problem to the metadata of the video being placed at the end of the .mp4 event file vs at the front. Executing "ffmpeg -i input.mp4 -codec copy -movflags faststart -f mp4 output.mp4" on any finished video clip will result in proper web playback as described in the first sentence. I've attempted to add the "faststart" parameter into the various ffmpeg parameters fields of Zoneminder but nothing seems to have any effect on the resulting event video.

Best I can narrow it down from apt logs and archived video, the change happened to Zoneminder sometime between April (I have a good clip from 4/9) and May (the oldest bad archived clip I have is on 5/2) of 2020, and I had updated between 1.34.8 to 1.34.10 between that date span.

**To Reproduce**
Steps to reproduce the behavior:
1. Record any video from a camera in H264 passthrough mode.
2. Attempt to play video event recorded in "H264 passthrough" mode in any web browser.
3. Observe that the file needs to download completely to be played properly.

**Expected behavior**
Events played back from "H264 passthrough" recorded cameras would start immediately, allow scrubbing, and basically stream the segments requested in a very pleasant way.

**Debug Logs**
```

connortechnology, 2020-08-06:
In 1.34.17 we now default to movflags=frag_keyframe which may fix this issue.  Also you can specify other values in the EncoderParams field so you should be able to experiment and try different values.

Tamahome-M, 2020-08-12:
I have the same problem, updating to the ZoneMinder v1.34.19 version did not fix the problem with the long start of playback in the browser.

connortechnology, 2020-08-12:
.19 just gives us the ability to change the setting.  So next step is to try different values.
So go to Monitor->Storage->Encoder Parameters
add an entry
movflags=frag_keyframe

You can try just abouthing from https://ffmpeg.org/ffmpeg-formats.html

Tamahome-M, 2020-08-13:
Unfortunately, adding "movflags=frag_keyframe" had no effect.
Also tried "movflags=frag_keyframe+faststart" and "movflags=faststart" ..

connortechnology, 2020-08-17:
The ongoing event SHOULD show in browser and play, but might not depending on movflags.  Please post debug zmc logs.

Tamahome-M, 2020-09-10:
```
Sep 10 14:42:03 nat zms_e162320[7768]: INF [zms_e162320] [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor is not part of your debug targets]
Sep 10 14:42:03 nat zms_e162320[7768]: ERR [zms_e162320] [Invalid frame id specified. 1 > 0]

```

incompetent-nerd, 2020-11-21:
> Unfortunately, adding "movflags=frag_keyframe" had no effect.
> Also tried "movflags=frag_keyframe+faststart" and "movflags=faststart" ..
> By the way, should the ongoing event show in the browser?

This worked for me: movflags=frag_custom+dash+delay_moov

Tamahome-M, 2020-11-21:
> This worked for me: movflags=frag_custom+dash+delay_moov

I tried your parameters, but they did not work for me, it even got worse.

okgo88, 2021-01-02:
> > Unfortunately, adding "movflags=frag_keyframe" had no effect.
> > Also tried "movflags=frag_keyframe+faststart" and "movflags=faststart" ..
> > By the way, should the ongoing event show in the browser?
> 
> This worked for me: movflags=frag_custom+dash+delay_moov

This also worked for me :)

closed 2021-01-02 by okgo88

dodexahedron, 2021-02-28:
> This worked for me: movflags=frag_custom+dash+delay_moov

Another upvote for this one, if anybody comes looking.

fennec622, 2021-03-01:
> > This worked for me: movflags=frag_custom+dash+delay_moov
> 
> Another upvote for this one, if anybody comes looking.

For chrome it's ok

But same problem on safari mac or ios with movflags=frag_custom+dash+delay_moov

Do you know another option ?

thanks

gk998, 2021-04-29:
> In 1.34.17 we now default to movflags=frag_keyframe which may fix this issue. Also you can specify other values in the EncoderParams field so you should be able to experiment and try different values.
As the issue is still present in iOS (Safari and zmNinja), can you tell us, which movflags have been used before 1.34.17?

incompetent-nerd, 2021-05-02:
I went into each monitor and under storage/optional encoder parameters, I added the following line:
**movflags=faststart**
So now I have only the above line and nothing else.

Now, all my devices (Chrome, Firefox, Edge, ZMNinja on Windows/IOS/Android) all work like before, i.e. playing begins immediately. One thing I have noticed is sometimes my event durations are not 10 minutes long. Sometimes they are 15 minutes, sometimes they are 2 minutes, etc... Not sure why.
I recall trying this setting a few months ago and it didn't do anything, so I'm not sure if anything was changed in the most recent builds or not.