ZoneMinder/zoneminder #2985: Monitor does not display correctly when "Rotate Right" is selected for "Orientation" in ZoneMinder

neillbell, 2020-07-21:
**Describe Your Environment**
- Version of ZoneMinder: 1.35.5
- How you installed ZoneMinder: master PPA/1.35.5~20200721160555-focal
- Full name and version of OS: Ubuntu 20.04 LTS

**Describe the bug**
Monitor does not display correctly when "Rotate Right" is selected for "Orientation" in ZoneMinder.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a new FFMPEG monitor with normal orientation
2. Click on the monitor name, the monitor displays correctly.
3. Click on "Console" to return to the main console.
4. Click on the camera address to bring up the monitor configuration screen.
5. Select the "Source" tab and set the monitor orientation to "Rotate Right"
6. Click the "Save" button.
7. When the monitor is up, click on its name.
8. The monitor displays incorrectly at any scale as shown below. The bottom half of the video frequently shows flashes of the non-rotated display.

[image]

connortechnology, 2020-07-29:
I think I fixed this.  Please confirm.

neillbell, 2020-07-30:
You did. I'm sorry that I didn't close this.

closed 2020-07-30 by connortechnology