ZoneMinder/zoneminder #2987: ZM fails to send email

weswitt, 2020-07-23:
**Describe Your Environment**
- Version of ZoneMinder: current release version
- How you installed ZoneMinder: Ubuntu PPA
- Full name and version of OS: 
Distributor ID: Ubuntu
Description:    Ubuntu 20.04 LTS
Release:        20.04
Codename:       focal
- Browser name and version (if this is an issue with the web interface): n/a

**Describe the bug**
ZM fails to send email for a filter event.

I have MSMTP properly configured and I can send email via the command line using MAIL, MSMTP and SENDMAIL.

When I attempt to send an email for an event from the filter page I get this error in the log: "Unable to send email: error closing /usr/lib/sendmail: (exit 19968)".

Jul 22 14:28:44 zoneminder zmfilter_4[1070]: INF [Creating notification email]
Jul 22 14:28:44 zoneminder zmfilter_4[1070]: INF [Sending notification email 'ZoneMinder: Alarm - Monitor-1-1 (26 - 16 18)']
Jul 22 14:28:44 zoneminder zmfilter_4[1070]: ERR [Unable to send email: error closing /usr/lib/sendmail: (exit 19968)]

Email options:
OPT_EMAIL: enabled
OPT_MESSAGE: enabled
NEW_MAIL_MODULES: enabled
SSMTP_MAIL: enabled
SSMTP_PATH: /usr/bin/msmtp
EMAIL_ADDRESS, MESSAGE_ADDRESS, FROM_EMAIL: all set to my email address

It seems that for some reason zm is using /usr/lib/sendmail instead of the configured email client. there are no entries in the msmtp log.

Furthermore i write a simple perl script that simulates what zm does in zmfilter.pl and it works perfectly:

#!/usr/bin/perl
use MIME::Lite;
use Net::SMTP;
my $from_address = 'foo@bar.com';
my $to_address = 'foo@bar.com';
my $subject = 'test msg';
my $message_body = "*** body ****";
$msg = MIME::Lite->new(
From => $from_address,
To => $to_address,
Subject => $subject,
Type =>'multipart/mixed'
);
$msg->attach(Type => 'TEXT', Data => $message_body);
$msg->send('sendmail', '/usr/bin/msmtp', $from_address);

knight-of-ni, 2020-07-23:
Have you tried the forums?
https://forums.zoneminder.com/viewtopic.php?t=25346

When testing sample code, you need to test with the web account user, rather than your own user account, which in your case is www-data.

closed 2020-07-31 by no-response[bot]

james-choncholas, 2020-08-03:
Same issue here