ZoneMinder/zoneminder #2988: Axis API V2 needs to be updated to allow " cameraId=<##> " to append to GET /axis-cgi/com/ptz.cgi

good4y0u, 2020-07-23:
**Describe Your Environment**
- Version of ZoneMinder release version v1.34.16.
- How you installed ZoneMinder Docker Container https://hub.docker.com/r/quantumobject/docker-zoneminder/ 
- Full name and version of OS
- Browser name and version 
Brave - Version 1.11.97 Chromium: 84.0.4147.89 (Official Build) (64-bit)
Firefox- 78.0.2 (64-bit)

**If the issue concerns a camera**
- Make and Model : conico (Axis PTZ ) 
- resolution 1080p
However I am connecting to TinyCam. Tiny cam IS able to contro the Conico PTZ. 

**Describe the bug**
In Control Axis API v2 the ?cameraid=### option does not work. 
Here is what the PTZ command looks like from tinycam GET /axis-cgi/com/ptz.cgi?cameraId=###&continuouspantiltmove=100,0

Here is what it looks like from zoneminder, note the lack of cameraID. When I add it to the URL it doesn't work, I need to change it in the PTZ config...I was hoping someone else already had working one.

GET /axis-cgi/com/ptz.cgi?rpan=-40

**To Reproduce**
Steps to reproduce the behavior:
1. setup tinycam app 
2. enable server
3. setup two ptz cameras
4. add tinycam server to zoneminder
6. ensure both cameras are showing up with their respective streams using the following format 
source: http://<ip>:8083/axis-cgi/mjpg/video.cgi?camera=<camera #>&user=<user>&pwd=<password>
Control-Type: <user>:<password>@192.168.1.157:8083

7. try to select a specific camera for PTZ
I tried with - control device, I Tried to add it to the end of the url, I tried setting Control Device to "/axis-cgi/com/ptz.cgi?cameraId=1" 

**Expected behavior**
I expect to be able to pass ?cameraId=1 to tinycam .
Tinycam log shows: 
GET /axis-cgi/com/ptz.cgi?rpan=-40
when it should say 
GET /axis-cgi/com/ptz.cgi**?cameraId=1**&rpan=-40

good4y0u, 2020-07-24:
> Thanks for opening your first issue here! Just a reminder, this forum is for Bug Reports only. Be sure to follow the issue template!

Thanks bot, this is in fact a bug report. API needs to look like the following, except as a perl script. note the /ptz.cgi?camera=%CAMERANO%  section 

`<request name="PtzMoveRelLeft">/axis-cgi/com/ptz.cgi?camera=%CAMERANO%&amp;continuouspantiltmove=-100,0</request>
`

good4y0u, 2020-09-27:
bump -- if someone knows how I can update this file myself and have it load with zoneminder I can do that. I tried editing the file however zoneminder would not recognize the new edits.

connortechnology, 2020-09-28:
you have to restart the control daemon to get it to pick up your changes.
sudo zmdc.pl restart /usr/bin/zmcontrol.pl --id 1