ZoneMinder/zoneminder #2991: Bootstrap 4.5 Punchlist

knight-of-ni, 2020-07-25:
ZoneMinder has been upgraded to Bootstrap 4.5. This was done to allow future upgrades to the zoneminder navigation bar.

This issue has been created to track formatting issues that have not yet been corrected:
- Button styling overrides are not applying
- Color states of the widgets of on the navigation bar are not applying (logs, /dev/shm, storage areas, etc)
- The dark skin has not been tested
- verify font sizes ???

If you notice something that looks different than it did before, please post it to this issue.

knight-of-ni, 2020-07-25:
Hey, hey, look at that... it thinks this is my first issue.

jonrub, 2020-07-25:
I guess I get below errors because of this?

[https://imgur.com/a/Qa29rEk](url)

knight-of-ni, 2020-07-25:
You need the latest build of master to fix that

knight-of-ni, 2020-07-26:
Navbar Widget colorization has been fixed in commit 6486f59
Please test!

bkjaya2020, 2020-07-26:
In Zoneminder 1.35.5~20200725145046-focal , the filters can't be saved.

knight-of-ni, 2020-07-26:
> In Zoneminder 1.35.5~20200725145046-focal , the filters can't be saved.

Working fine here. You didn't include any error message or screenshot, so what lead you to believe this is related to the recent bootstrap upgrade?

bkjaya2020, 2020-07-26:
> > In Zoneminder 1.35.5~20200725145046-focal , the filters can't be saved.
> 
> Working fine here. You didn't include any error message or screenshot. What lead you to believe this is related to the recent bootstrap upgrade?

In the filter , the following  conditional statements are not saved. 
https://drive.google.com/file/d/1i86XWjHSINIZq0YQnCNT59DGW9apa2Xc/view?usp=sharing

The error log is attached.

https://drive.google.com/file/d/1pZXYt_LtLeEGkTeToRqjfvteNBuDAVNa/view?usp=sharing

knight-of-ni, 2020-07-26:
I don't have permission to view those links.

bkjaya2020, 2020-07-26:
> I don't have permission to view those links.

Sorry for the mistake
[https://www.dropbox.com/s/jhq1vdy8osx9b5p/conditionalstatement?dl=0](url)

[https://www.dropbox.com/s/jhq1vdy8osx9b5p/conditionalstatement?dl=0](url)

knight-of-ni, 2020-07-26:
Those download links redirect back to github, strangely. After reading your post in the forums, I was able understand and duplicate the problem. The problem was caused by yesterday's changes to Object.php.
I've let Isaac know. Since this issue is not related to the Bootstrap upgrade, please keep further discussion here:
https://forums.zoneminder.com/viewtopic.php?f=42&p=116674#p116674

bkjaya2020, 2020-07-27:
Thank you Sir

UtechtDustin, 2020-08-16:
> Table changes are not being remembered - cookie extension not working
> UPDATE: Cookie size it too big. See wenzhixin/bootstrap-table#5259

Will be fixed in the next bootstrap-table version (see https://github.com/wenzhixin/bootstrap-table/pull/5260 and https://github.com/wenzhixin/bootstrap-table/issues/5259)

knight-of-ni, 2020-08-20:
Fix black table text

[image]

knight-of-ni, 2020-09-16:
Sample screenshot showing frames with with > 25 frames
[image]

knight-of-ni, 2020-10-25:
@mnoorenberghe 

Server side pagination for the Events view has just been pushed to master. This fixes the issue with the total number of records limited to WEB_EVENTS_PER_PAGE, for that view. PLEASE TEST. 

Server side pagination for the Frames view is the roadmap as well, but we need to test the pagination in the Events view further before proceeding.

knight-of-ni, 2021-01-24:
tasks are complete. closing...

closed 2021-01-24 by knight-of-ni