ZoneMinder/zoneminder #2996: Log: Downgrade to INFO "Unable to send raw frame 1: Broken Pipe"

PrplHaz4, 2020-07-29:
**Describe Your Environment**
- Version of ZoneMinder 1.34.16
- How you installed ZoneMinder Docker
- Browser name and version: Firefox 78.0.2

**Describe the bug**
Currently these messages are logged as ERRors.

This forum thread indicates they can be safely ignored:
https://forums.zoneminder.com/viewtopic.php?t=26109

As does this article:
https://mattionline.de/zoneminder-error-meanings/

**To Reproduce**
Steps to reproduce the behavior:
1. Open a live stream from the web console
2. Close the window containing the live stream
3. See the error in the log

**Expected behavior**
INFO message is logged instead of ERROR

**Debug Logs**
```
2020-07-29 11:42:09 | zms_e122678 |   | 30645 | ERR | Unable to send raw frame 1: Broken pipe | zm_eventstream.cpp | 816
```

connortechnology, 2020-07-29:
Agreed.  Fix is queued for 1.34.17.

closed 2020-07-29 by connortechnology

hdlineage, 2020-12-22:
I'm still getting this issue on v1.34.22 ( The stable official docker image)
Can you please confirm if this fix is reflected on the docker version?

Hawk84r, 2021-07-06:
I'm on on v1.34.22 not docker and can confirm it is still on ERRor level.

connortechnology, 2021-07-07:
You are correct, I missed getting it into 1.34.  I have now pushed it to the release-1.34 branch which means it will be in 1.34.27 when it gets released.

commit b1054a7d (2020-07-29): bail earlier if writing content-length fails. Dropped connections aren't really errors, make them Info.  Fixes #2996
commit b38ee562 (2021-07-07): Cherry pick b1054a7 which quiets logs when quitting due to unable to write frame image. Fixes #2996