ZoneMinder/zoneminder #2997: MYSQL database corrupted (every time)

Knowledge91, 2020-07-29:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.32
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
I use the quantumObject or dlandon docker image
- Full name and version of OS
Ubuntu 20.04
- Browser name and version (if this is an issue with the web interface)
Tried on current versions of Safari, Chrome and Firefox

**Describe the bug**
Zoneminder crashes my MySQL/ MariaDB database every time.

**To Reproduce**
Steps to reproduce the behavior:
- Run MySQL 5.7 or MariaDB 10.3.23 image
- Run dlandon or quantumObject zoneminder image
- Add 3 FFmpeg cameras and one libvlc camera
- wait for some hours 

**Expected behavior**
Database does not get corrupted.

**Debug Logs**
```
2020-07-29 21:51:58+02:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 5.7.31-1debian10 started.
2020-07-29 21:51:58+02:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'
2020-07-29 21:51:58+02:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 5.7.31-1debian10 started.
2020-07-29T19:51:58.998950Z 0 [Warning] TIMESTAMP with implicit DEFAULT value is deprecated. Please use --explicit_defaults_for_timestamp server option (see documentation for more details).
2020-07-29T19:51:59.002496Z 0 [Note] mysqld (mysqld 5.7.31) starting as process 1 ...
2020-07-29T19:51:59.011266Z 0 [Note] InnoDB: PUNCH HOLE support available
2020-07-29T19:51:59.011292Z 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2020-07-29T19:51:59.011299Z 0 [Note] InnoDB: Uses event mutexes
2020-07-29T19:51:59.011305Z 0 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2020-07-29T19:51:59.011310Z 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-07-29T19:51:59.011315Z 0 [Note] InnoDB: Using Linux native AIO
2020-07-29T19:51:59.011669Z 0 [Note] InnoDB: Number of pools: 1
2020-07-29T19:51:59.011854Z 0 [Note] InnoDB: Using CPU crc32 instructions
2020-07-29T19:51:59.014881Z 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2020-07-29T19:51:59.028013Z 0 [Note] InnoDB: Completed initialization of buffer pool
2020-07-29T19:51:59.030652Z 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-07-29T19:51:59.060407Z 0 [Note] InnoDB: Highest supported file format is Barracuda.
2020-07-29T19:51:59.063639Z 0 [Note] InnoDB: Log scan progressed past the checkpoint lsn 26596817
2020-07-29T19:51:59.063685Z 0 [Note] InnoDB: Doing recovery: scanned up to log sequence number 26616041
2020-07-29T19:51:59.071842Z 0 [Note] InnoDB: Database was not shutdown normally!
2020-07-29T19:51:59.071867Z 0 [Note] InnoDB: Starting crash recovery.
2020-07-29T19:51:59.134261Z 0 [Note] InnoDB: Starting an apply batch of log records to the database...
InnoDB: Progress in percent: 30 31 32 33 34 35 2020-07-29T19:51:59.134988Z 0 [ERROR] [FATAL] InnoDB: is_short 0, info_and_status_bits 0, offset 9055, o_offset 9, mismatch index 18446744073709551574, end_seg_len 90 parsed len 4
36 2020-07-29 21:51:59 0x7fa885ceb700  InnoDB: Assertion failure in thread 140361776150272 in file ut0ut.cc line 918
InnoDB: We intentionally generate a memory trap.
InnoDB: Submit a detailed bug report to http://bugs.mysql.com.
InnoDB: If you get repeated assertion failures or crashes, even
InnoDB: immediately after the mysqld startup, there may be
InnoDB: corruption in the InnoDB tablespace. Please refer to
InnoDB: http://dev.mysql.com/doc/refman/5.7/en/forcing-innodb-recovery.html
InnoDB: about forcing recovery.
19:51:59 UTC - mysqld got signal 6 ;
This could be because you hit a bug. It is also possible that this binary
or one of the libraries it was linked against is corrupt, improperly built,
or misconfigured. This error can also be caused by malfunctioning hardware.
Attempting to collect some information that could help diagnose the problem.
As this is a crash and something is definitely wrong, the information
collection process might fail.

key_buffer_size=8388608
read_buffer_size=131072
max_used_connections=0
max_threads=151
thread_count=0
connection_count=0
It is possible that mysqld could use up to 
key_buffer_size + (read_buffer_size + sort_buffer_size)*max_threads = 68196 K  bytes of memory
Hope that's ok; if not, decrease some variables in the equation.

Thread pointer: 0x0
Attempting backtrace. You can use the following information to find out
where mysqld died. If you see no messages after this, something went
terribly wrong...
37 38 stack_bottom = 0 thread_stack 0x40000
39 40 41 42 43 44 45 46 Fatal signal 11 while backtracing
```

connortechnology, 2020-07-29:
ZM cannot crash mysql.  That would be a bug in mysql.  It is more likely that your machine is unstable.

Further ZM 1.32 is very old and is not recommended for use.

Please upgrade to 1.34.

closed 2020-07-29 by connortechnology

Knowledge91, 2020-07-30:
I have 3 more databases running stable on the same Server having no problems at all.
I spend many hours on trying to figure out what the problem is. They only thing in common is zoneminder.
Having wasted enough time I decided to develop a proper solution. Thanks for nothing.