ZoneMinder/zoneminder #2998: Its not possible to remove a selected monitor

connectnet, 2020-07-30:
Hi there!
There is a small bug in Vers. 1.35.5 because its not possible to remove a selected monitor. E.ex. here Monitor-1.
Proof: Screenshot says more than 1000 Words. ;-)

[image]

Thanks for fixing in advance! :-)

connortechnology, 2020-07-30:
What do you mean remove?  What are you trying to do?

knight-of-ni, 2020-07-30:
It's working fine for me. Just added, then removed a monitor from the web console. Running a build from today.

connectnet, 2020-07-31:
You can add one or more monitors to decrease the results but you can't remove selected monitors from list.
Version: 1.35.5

connectnet, 2020-08-01:
Fixed by update. Thank you!

closed 2020-08-01 by connectnet

jeroenst, 2022-07-12:
Same problem with v1.36.4

I get these errors:

`
Date/Time,Component,Server,PID,Level,Message,File,Line
"2022-07-12 13:16:04",web_php,,11901,ERR,"Unable to load ZM\Event record for Id=402275",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,2462,ERR,"Unable to load ZM\Event record for Id=408494",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,918,ERR,"Unable to load ZM\Event record for Id=401913",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,448,ERR,"Unable to load ZM\Event record for Id=407466",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,938,ERR,"Unable to load ZM\Event record for Id=408348",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,681,ERR,"Unable to load ZM\Event record for Id=402033",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,680,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,931,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,2475,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,7682,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,447,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,6310,ERR,"Unable to load ZM\Event record for Id=402087",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,2475,ERR,"Unable to load ZM\Event record for Id=407845",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,447,ERR,"Unable to load ZM\Event record for Id=402513",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,931,ERR,"Unable to load ZM\Event record for Id=402525",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,680,ERR,"Unable to load ZM\Event record for Id=402020",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,7682,ERR,"Unable to load ZM\Event record for Id=401956",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,681,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,11901,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,918,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,448,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,938,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,2462,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,6310,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,918,ERR,"Unable to load ZM\Event record for Id=401911",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,681,ERR,"Unable to load ZM\Event record for Id=402029",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,2462,ERR,"Unable to load ZM\Event record for Id=408490",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,11901,ERR,"Unable to load ZM\Event record for Id=402273",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,448,ERR,"Unable to load ZM\Event record for Id=407461",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,938,ERR,"Unable to load ZM\Event record for Id=408344",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,447,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,6310,ERR,"Unable to load ZM\Event record for Id=402081",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,931,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,7682,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,2475,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,680,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:16:04",web_php,,2475,ERR,"Unable to load ZM\Event record for Id=407836",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,931,ERR,"Unable to load ZM\Event record for Id=402521",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,447,ERR,"Unable to load ZM\Event record for Id=402510",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:16:04",web_php,,680,ERR,"Unable to load ZM\Event record for Id=402018",/usr/share/zoneminder/www/includes/Object.php,20
[... cut ...]
"2022-07-12 13:15:59",web_php,,6310,ERR,"Event delete on event with empty Id",/usr/share/zoneminder/www/includes/Event.php,135
"2022-07-12 13:15:59",web_php,,2462,ERR,"Unable to load ZM\Event record for Id=408468",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:15:59",web_php,,918,ERR,"Unable to load ZM\Event record for Id=401883",/usr/share/zoneminder/www/includes/Object.php,20
"2022-07-12 13:15:59",web_php,,7682,ERR,"Unable to load ZM\Event record for Id=401913",/usr/share/zoneminder/www/includes/Object.php,20
`