ZoneMinder/zoneminder #2999: various mistakes in git sources after 24.7.2020

zd59, 2020-07-30:
**Describe Your Environment**
Zoneminder GIT source after 24.7.2020 14:57 GMT+2
packages created with updated script from slackbuilds.org
Slackware64-current updated 30.07.2020 - two machines, one as VirtualBox guest, the other is physical
Chrome or Mozilla Firefox

**Describe the bug**
Zoneminder compiled from GIT after above mentioned have those failures:
1. status change button misbehave:
 - sometimes not active (do not offer drop-down menu to select Stop/Start..) and click on it sometimes freeze the page (all buttons become unclickable).
 - when ZM is in a running state and managed in Chrome browser the VirtualBox Linux machine crashes with "can not access memory.. " failure. Physical machine do not crash, only the state button is not active.

2. Console Cameras status for instance "Capturing" is black on "almost black" background in a Dark theme - amost not visible

3. status button is too big

**To Reproduce**
Nothing special, just check it out.

**Expected behavior**
Fix  the above

**Debug Logs**
When crash, there are no errors reported.

```

Carbenium, 2021-05-16:
@zd59 Are these things still valid with 1.36.0?

connortechnology, 2021-05-16:
status button still needs an overhaul.  It's bad UI.  The whole states concept is half-implemented..

dark theme needs love.  I improved it a bit, but I'm no UI expert or graphic designed. 

These should be put in their own issues though and tagged as todo or wishlist

zd59, 2021-05-18:
> @zd59 Are these things still valid with 1.36.0?
Yesterday I've installed 1.36.0 to Virtual machine Slackware64-current.
Regarding status button now it's all OK, works & looks fine.
Visibility is stil a problem on main menu - all grey is hard to read:
[image]
Dark theme - help invisible:
[image]
Camera link speed, as main menu, all grew is hard to read:
[image]
And switching to Classic CSS enlarge all fonts everywhere
All other: I must congratulate to user interface it is excellent and even better in time.

closed 2025-12-27 by SteveGilvarry