ZoneMinder/zoneminder #3004: Manpage should not be prefixed with zoneminder-

marillat, 2020-08-05:
Manpage should follow the binary name.

A patch is proposed in PR https://github.com/ZoneMinder/zoneminder/pull/3003

knight-of-ni, 2020-08-05:
I don't see any reason to change this. It doesn't affect the user experience. Typing `man zma`, `man zmtrack.pl`, etc from the commandline produces the desired manpage.

marillat, 2020-08-06:
The common usage is to call the filename as the binary name, or function name like perl manpage.

FYI Debian remove "zoneminder-" to your manpage files name.

https://packages.debian.org/sid/amd64/zoneminder/filelist

knight-of-ni, 2020-08-06:
That isn't necessarily true, and one doesn't have to look far to see counter examples.

There man pages do not match the binary name:
```
$ ls /usr/share/man/man1/{ffmpeg*,git*}
/usr/share/man/man1/ffmpeg.1.gz
/usr/share/man/man1/ffmpeg-all.1.gz
/usr/share/man/man1/ffmpeg-bitstream-filters.1.gz
/usr/share/man/man1/ffmpeg-codecs.1.gz
/usr/share/man/man1/ffmpeg-devices.1.gz
/usr/share/man/man1/ffmpeg-filters.1.gz
/usr/share/man/man1/ffmpeg-formats.1.gz
/usr/share/man/man1/ffmpeg-protocols.1.gz
/usr/share/man/man1/ffmpeg-resampler.1.gz
/usr/share/man/man1/ffmpeg-scaler.1.gz
/usr/share/man/man1/ffmpeg-utils.1.gz
/usr/share/man/man1/git.1.gz
/usr/share/man/man1/git-add.1.gz
/usr/share/man/man1/git-am.1.gz
/usr/share/man/man1/git-annotate.1.gz
/usr/share/man/man1/git-apply.1.gz
/usr/share/man/man1/git-archive.1.gz
/usr/share/man/man1/git-bisect.1.gz
/usr/share/man/man1/git-blame.1.gz
/usr/share/man/man1/git-branch.1.gz
/usr/share/man/man1/git-bundle.1.gz
/usr/share/man/man1/git-cat-file.1.gz
/usr/share/man/man1/git-check-attr.1.gz
/usr/share/man/man1/git-check-ignore.1.gz
/usr/share/man/man1/git-check-mailmap.1.gz
/usr/share/man/man1/git-checkout.1.gz
/usr/share/man/man1/git-checkout-index.1.gz
/usr/share/man/man1/git-check-ref-format.1.gz
/usr/share/man/man1/git-cherry.1.gz
/usr/share/man/man1/git-cherry-pick.1.gz
/usr/share/man/man1/git-clean.1.gz
/usr/share/man/man1/git-clone.1.gz
/usr/share/man/man1/git-column.1.gz
/usr/share/man/man1/git-commit.1.gz
/usr/share/man/man1/git-commit-graph.1.gz
/usr/share/man/man1/git-commit-tree.1.gz
/usr/share/man/man1/git-config.1.gz
/usr/share/man/man1/git-count-objects.1.gz
/usr/share/man/man1/git-credential.1.gz
/usr/share/man/man1/git-credential-cache.1.gz
/usr/share/man/man1/git-credential-cache--daemon.1.gz
/usr/share/man/man1/git-credential-store.1.gz
/usr/share/man/man1/git-daemon.1.gz
/usr/share/man/man1/git-deborig.1.gz
/usr/share/man/man1/git-describe.1.gz
/usr/share/man/man1/git-diff.1.gz
/usr/share/man/man1/git-diff-files.1.gz
/usr/share/man/man1/git-diff-index.1.gz
/usr/share/man/man1/git-difftool.1.gz
/usr/share/man/man1/git-diff-tree.1.gz
/usr/share/man/man1/git-fast-export.1.gz
/usr/share/man/man1/git-fast-import.1.gz
/usr/share/man/man1/git-fetch.1.gz
/usr/share/man/man1/git-fetch-pack.1.gz
/usr/share/man/man1/git-filter-branch.1.gz
/usr/share/man/man1/git-fmt-merge-msg.1.gz
/usr/share/man/man1/git-for-each-ref.1.gz
/usr/share/man/man1/git-format-patch.1.gz
/usr/share/man/man1/git-fsck.1.gz
/usr/share/man/man1/git-fsck-objects.1.gz
/usr/share/man/man1/git-gc.1.gz
/usr/share/man/man1/git-get-tar-commit-id.1.gz
/usr/share/man/man1/git-grep.1.gz
/usr/share/man/man1/git-hash-object.1.gz
/usr/share/man/man1/git-help.1.gz
[... 70 lines cut ...]
/usr/share/man/man1/git-stash.1.gz
/usr/share/man/man1/git-status.1.gz
/usr/share/man/man1/git-stripspace.1.gz
/usr/share/man/man1/git-submodule.1.gz
/usr/share/man/man1/git-subtree.1.gz
/usr/share/man/man1/git-switch.1.gz
/usr/share/man/man1/git-symbolic-ref.1.gz
/usr/share/man/man1/git-tag.1.gz
/usr/share/man/man1/git-unpack-file.1.gz
/usr/share/man/man1/git-unpack-objects.1.gz
/usr/share/man/man1/git-update-index.1.gz
/usr/share/man/man1/git-update-ref.1.gz
/usr/share/man/man1/git-update-server-info.1.gz
/usr/share/man/man1/git-upload-archive.1.gz
/usr/share/man/man1/git-upload-pack.1.gz
/usr/share/man/man1/git-var.1.gz
/usr/share/man/man1/git-verify-commit.1.gz
/usr/share/man/man1/git-verify-pack.1.gz
/usr/share/man/man1/git-verify-tag.1.gz
/usr/share/man/man1/gitweb.1.gz
/usr/share/man/man1/git-web--browse.1.gz
/usr/share/man/man1/git-whatchanged.1.gz
/usr/share/man/man1/git-worktree.1.gz
/usr/share/man/man1/git-write-tree.1.gz
```

knight-of-ni, 2020-08-06:
Is the current naming convention actually causing a problem with Debian packaging or is this simply someone's opinion on how something should look? So far it seems to be the latter.

marillat, 2020-08-06:
> That isn't necessarily true, and one doesn't have to look far to see counter examples.
> 
> These man pages do not match the binary name:
> 
> ```
> $ ls /usr/share/man/man1/{ffmpeg*,git*}
> /usr/share/man/man1/ffmpeg.1.gz
> /usr/share/man/man1/ffmpeg-all.1.gz
>```
Git and ffmpeg manpages have been split for readability.

> By the way the debian package missed a few:
> 
> ```
> /usr/share/man/man8/zoneminder-zma.8.gz
> /usr/share/man/man8/zoneminder-zmc.8.gz
> /usr/share/man/man8/zoneminder-zmu.8.gz
> ```

Hard for a debian maintainer to find a manpage who deosn't match the binary name. :)

> Is the current naming convention actually causing a problem with Debian packaging or is this simply someone's opinion on how something should look? So far it seems to be the latter, but I am willing to change it if this is causing some kind of real problem.

marillat, 2020-08-06:
> Is the current naming convention actually causing a problem with Debian packaging or is this simply someone's opinion on how something should look? So far it seems to be the latter, but I am willing to change it if this is causing some kind of real problem.

From lintian (Debian package checker)

```
 Note that though the **man** program has the capability to check for
 several program names in the NAMES section, each of these programs
 should have its own manual page (a symbolic link to the appropriate
 manual page is sufficient) because other manual page viewers such as
 xman or tkman don't support this.
 .
 If the name of the manual page differs from the binary by case, **man**
 may be able to find it anyway; however, it is still best practice to match
 the exact capitalization of the executable in the manual page.
```

knight-of-ni, 2020-08-06:
There we go. That's a good reason. Debian lintian check is flagging the build with a warning or error. There is a similar pod2man statement that also needs to be changed under /src. Can you add that to your pr? I want this change to be consistent.

marillat, 2020-08-07:
> There we go. That's a good reason. Debian lintian check is flagging the build with a warning or error. There is a similar pod2man statement that also needs to be changed under /src. Can you add that to your pr? I want this change to be consistent.

I added src/CMakeLists.txt to PR #30003

knight-of-ni, 2020-08-07:
Closed via PR #3003

closed 2020-08-07 by knight-of-ni