ZoneMinder/zoneminder #3005: 1.34.17 Unusable image when monitor orientation is set to rotated

Fredrick-III, 2020-08-06:
**Describe Your Environment**
- ZoneMinder release 1.34.17
- Installed from PPA
- Ubuntu 20.04.1 and 18.04.4
- Latest Chrome and Firefox

**Describe the bug**
When a monitor's orientation is set to rotate left or rotate right, the monitor is unusable and appears as a bunch of lines, occasionally half of the correct image will appear. This appeared on 2 servers (Ubuntu 20.04 and 18.04) after the upgrade to 1.34.17 from the PPA.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit camera source
2. Go to Source tab
3. Set orientation to rotate left or rotate right
4. Look at monitor and see unusable image

**Expected behavior**
Rotated monitors should appear clear, just rotated left or right.

[image]

connortechnology, 2020-08-06:
Should be fixed in 1.34.18.

Fredrick-III, 2020-08-06:
Just updated to 1.34.18 on both servers and everything works. Thanks!

closed 2020-08-06 by Fredrick-III