ZoneMinder/zoneminder #3007: DiskSpace only calculated on the first refresh of Events

MarkBarbieri, 2020-08-07:
**Describe Your Environment**
- New install of Master on debian Buster (v10.5) from deb https://zmrepo.zoneminder.com/debian/master buster/
- 1.35.5~20200806.315
- Microsoft Edge Chromium Browser

**Describe the bug**
The Events listing does not display the correct DiskSpace for each recording when set to record continuously (MOCORD).  DiskSpace is calculated the first time the UI updates the event list, and the size is never updated again.

**To Reproduce**
Steps to reproduce the behavior:
1. Set to MOCORD
2. Click on Events listing
3. Repeatedly refresh the UI, note that the DiskSpace updates on the first refresh, but never again

**Expected behavior**
DiskSpace should be recalculated on each refresh

connortechnology, 2020-08-07:
True.  This should only affect in-progress events though.  Once an event is finished it's size shouldn't change significantly so there is no point updating it.

Should be a simple fix.

MarkBarbieri, 2020-08-07:
> True. This should only affect in-progress events though. Once an event is finished it's size shouldn't change significantly so there is no point updating it.
> 
> Should be a simple fix.

True, but once the in-progress recording completes the size is not updated, so the completed events are also incorrect.

commit eff96c5f (2020-08-07): Only save the DiskSpace on completed events. Fixes #3007
commit da757d07 (2020-08-07): Only save the DiskSpace on completed events. Fixes #3007
closed 2020-08-07 by connortechnology