ZoneMinder/zoneminder #3011: not scale in montage v1.34.16

SaraDark, 2020-08-08:
Zoneminder v1.34.16 in Montage not set scale parametr on url, eg. &scale=
This is very big problem in 25+ monitors, server CPU very high when must send full hd all monitors, in older version scale working fine and CPU work fine.

eg.
/cgi-bin-zm/nph-zms?width=0%25&height=auto&scale=&mode=jpeg&maxfps=3&monitor=2&auth=8d28faca194c1d40b00cf0c9f5ba43b3&connkey=673556&rand=1596931024

OK example (scale diffrent by monitor resolution on vieving devices).:
/cgi-bin-zm/nph-zms?width=0%25&height=auto&scale=40&mode=jpeg&maxfps=3&monitor=2&auth=8d28faca194c1d40b00cf0c9f5ba43b3&connkey=673556&rand=1596931024

connortechnology, 2020-08-13:
Can you confirm that this is fixes in 1.34.19?

SaraDark, 2020-08-14:
Sorry but at this moment a cannot check, i reverted to previous version, this is machine in production and downtime is critical issue :/

closed 2020-09-24 by connortechnology