ZoneMinder/zoneminder #3014: Various undefined constant and zmc not running after master update

rhaamo, 2020-08-12:
- Version: master (zoneminder_1.35.5~20200812172754-focal_amd64.deb)
- Via: do_debian_package

Since the update `(1.35.5~20200812172754-focal) over (1.35.5~20200627074706-focal)` I have various errors displayed in web ui:

```
Warning: Use of undefined constant ZM_WEB_NAVBAR_TYPE - assumed 'ZM_WEB_NAVBAR_TYPE' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/skins/classic/includes/functions.php on line 239
Warning: Use of undefined constant ZM_HOME_ABOUT - assumed 'ZM_HOME_ABOUT' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/skins/classic/includes/functions.php on line 606
```

The header is weirdly broken too, probably because of the error.
[image]

The log file `/var/log/zm/web_php.log` does seems to have lots of "useless" informations in it now:
```
08/12/20 18:12:13.621127 web_php[3002902].DBG [192.168.10.1] [daemonControl /usr/bin/zmdc.pl start zmc -m 3] at /usr/share/zoneminder/www/includes/functions.php line 813
08/12/20 18:12:13.918141 web_php[3002902].DBG [192.168.10.1] [daemonControl /usr/bin/zmdc.pl start zma -m 3] at /usr/share/zoneminder/www/includes/functions.php line 813
08/12/20 18:12:14.744995 web_php[2895824].DBG [192.168.10.1] [<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN" "DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml"><head>
<style type="text/css">
body {background-color: #fff; color: #222; font-family: sans-serif;}
pre {margin: 0; font-family: monospace;}
a:link {color: #009; text-decoration: none; background-color: #fff;}
a:hover {text-decoration: underline;}
table {border-collapse: collapse; border: 0; width: 934px; box-shadow: 1px 2px 3px #ccc;}
.center {text-align: center;}
.center table {margin: 1em auto; text-align: left;}
.center th {text-align: center !important;}
td, th {border: 1px solid #666; font-size: 75%; vertical-align: baseline; padding: 4px 5px;}
th {position: sticky; top: 0; background: inherit;}
h1 {font-size: 150%;}
h2 {font-size: 125%;}
.p {text-align: left;}
.e {background-color: #ccf; width: 300px; font-weight: bold;}
.h {background-color: #99c; font-weight: bold;}
.v {background-color: #ddd; max-width: 300px; overflow-x: auto; word-wrap: break-word;}
.v i {color: #999;}
img {float: right; border: 0;}
hr {width: 934px; background-color: #ccc; border: 0; height: 1px;}
</style>
<title>PHP 7.4.3 - phpinfo()</title><meta name="ROBOTS" content="NOINDEX,NOFOLLOW,NOARCHIVE" /></head>
<body><div class="center">
<h2>PHP Variables</h2>
<table>
<tr class="h"><th>Variable</th><th>Value</th></tr>
<tr><td class="e">$_REQUEST['view']</td><td class="v">console</td></tr>
<tr><td class="e">$_GET['view']</td><td class="v">console</td></tr>
<tr><td class="e">$_COOKIE['zmSkin']</td><td class="v">classic</td></tr>
<tr><td class="e">$_COOKIE['zmCSS']</td><td class="v">base</td></tr>
<tr><td class="e">$_COOKIE['zmBandwidth']</td><td class="v">medium</td></tr>
<tr><td class="e">$_COOKIE['zmEventRate']</td><td class="v">100</td></tr>
...
```

`zmtrigger.log` does have lots of:
```
08/12/2020 18:14:44.781429 zmtrigger[3093801].ERR [ZoneMinder::Memory::Mapped:90] [Memory map file '/dev/shm/zm.mmap.3' does not exist in zmMemAttach.  zmc might not be running.]
```

and indeed all my monitors are red and zmc isn't running (

Database seems up to date:
```
root@fenouil:/etc/zm# zmupdate.pl --interactive

Database already at version 1.35.5, update skipped.
```

Is something needed for that upgrade ?

connortechnology, 2020-08-12:
You haven't updated your db.  run zmupdate.pl -f

connortechnology, 2020-08-12:
The structure is ok, but the contents of the config table are not.

rhaamo, 2020-08-12:
my bad I always used only `zmupdate.pl --interactive` without the `-f` flag.

That fixed zmc and display errors, thanks !

However there is still some html in `zm_php.log` but I think this is another issue, or expected ?

connortechnology, 2020-08-13:
what is the html in zm_php.log?

closed 2020-08-17 by connortechnology