ZoneMinder/zoneminder #3016: ffmpeg vs ustreamer

Tampa, 2020-08-19:
I personally been using zoneminder to make sure shipping companies don't just leave packages in front of my door unless told to so I only use a single camera, but on the system it is connected to zoneminder consumes pretty much 90% of all resources, granted it is only a dual core with very limited memory, but I am still looking to increase performance. After having exhausted all manners of tuning zoneminder I started to look towards other options.

Then I stumbled upon an article about TinyPilot which used ffmpeg to render kvm over ip, but evidently the resource usage and subsequent latency issues were problematic. The apparent solution was called ustreamer, which not only reduced latency, but also offered lower system overhead for their usecase. Now this may be different from what zoneminder requires, but maybe parts of the technology can still be useful to reduce the resource usage of zoneminder.

https://github.com/pikvm/ustreamer
https://mtlynch.io/tinypilot/#ustreamer-a-super-fast-video-streamer

Simply wanted to put that out there since I couldn't find an issue ticket mentioning ustreamer.

PrplHaz4, 2020-08-19:
Unfortunately I think what’s eating resources is the motion detection not the streaming. I think the two most impactful changes are to reduce  camera resolution or reduce framerate on the camera.The docs have a fairly lengthy guide for tweaking performance.

Tampa, 2020-08-19:
I run record on the camera rather than modect, because the latter would not work at all due to the lower quality camera and lack of system resources.

connortechnology, 2020-08-19:
If you are recording from a v4l source you should probably be using our local camera monitor type.  

In any event, while ustreamer is a nice thing to know about this is a place for bug reports and hence inappropriate.

closed 2020-08-19 by connortechnology

Tampa, 2020-08-19:
Then where should I put this exactly?

connortechnology, 2020-08-19:
You can discuss it in the forums if you want.