ZoneMinder/zoneminder #3022: CI:

Carbenium, 2020-08-24:
**Describe Your Environment**
- Travis CI

**Describe the bug**
It is currently a bit unclear which distro config (`distros/`) should be used for which CI job or the naming of the folders is unclear.

**Current configs in use:**

**Debian**
* Jessie -> `ubuntu1604`
* Stretch -> `ubuntu1604`
* Buster -> `ubuntu1604`

**Ubuntu**
* Trusty -> `ubuntu1204`
* Xenial -> `ubuntu1604`
* Bionic -> `ubuntu1604`
* Disco -> `ubuntu1604`
* Focal -> `ubuntu1604`

**Devuan (not included in CI; just for completeness)**
* Beowulf -> `beowulf`

**Fedora/EL** -> `redhat`

From this listing one can see that `debian`, ``ubuntu1504_cmake_split_packages` and `ubunt2004` are unused. Is this expected or is something mixed up here?
Are opensuse build provided?

Carbenium, 2020-08-24:
With 2e4baba740341bd6c702459e79a42a415b681948 the list is changed to:

**Debian**
* Jessie -> `ubuntu1604` (should be deprecated; #3020)
* Stretch -> `ubuntu1604`
* Buster -> `ubuntu2004`

**Ubuntu**
* Trusty -> deprecated
* Xenial -> `ubuntu1604`
* Bionic -> `ubuntu1604`
* Disco -> `ubuntu1604` (should be deprecated; #3020)
* Focal -> `ubuntu2004`

Usage of `debian` and `ubuntu1504_cmake_split_packages` remain unclear.

connortechnology, 2020-08-25:
debian is for really old debian like wheezy.  ubuntu1504_cmake_split_packages is for reference because someone did the work to split up ZM into multiple packages and I would like to revisit it someday.

I havn't deprecated jessie because I have a paying client running jessie.  When I get them upgraded we can begin discussing it.  Plus we should consult our telemetry to see what people are using.

connortechnology, 2021-05-10:
@Carbenium what are you looking for in this issue?  What needs to be done to resolve it?

SteveGilvarry, 2026-06-12:
Triaging the issue backlog. This documented Travis CI distro configs; CI moved to GitHub Actions years ago. Closing.