ZoneMinder/zoneminder #3029: The console will not display more than WEB_EVENTS_PER_PAGE events

neillbell, 2020-08-26:
**Describe Your Environment**
- Version of ZoneMinder: v1.35.6 (1.35.6~20200825114814-bionic)
- How you installed ZoneMinder: PPA zoneminder-master
- Full name and version of OS: Ubuntu 18.04.5 LTS
- Browser name and version: Chromium 84.0.4147.105

**Describe the bug**
The console will not display more than WEB_EVENTS_PER_PAGE events no matter how many events are currently available.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the ZoneMinder console and click on one of the event count links that shows a number that is greater than WEB_EVENTS_PER_PAGE.
2. By default, the display will show a paged listing with a notice at the bottom of the page that it is displaying '1 to 10 of WEB_EVENTS_PER_PAGE' rows.
3. Click on the the 'Hide/Show pagination' button.
4. The display will show a non-paged listing of WEB_EVENTS_PER_PAGE rows.

**Expected behavior**
In paged mode, the console should allow the user to page through all of the available events rather than the first WEB_EVENTS_PER_PAGE events. In non-paged mode the console should display all of the available events rather than this first WEB_EVENTS_PER_PAGE events.

If the number of events available is greater than WEB_EVENTS_PER_PAGE, only the first WEB_EVENTS_PER_PAGE are accessible.

knight-of-ni, 2020-08-26:
We've already explain the situation in the forums and in slack, and we've also asked not to spawn multiple issues related to the new UI.

knight-of-ni, 2020-08-26:
This is already being tracked here: https://github.com/ZoneMinder/zoneminder/issues/2991

closed 2020-08-26 by knight-of-ni