ZoneMinder/zoneminder #3033: "Back" link doesn't work when viewing events.

jstenback, 2020-09-04:
**Describe Your Environment**
- Version of ZoneMinder: 1.35.6.r15917.dd5e34219-1
- How you installed ZoneMinder: Arch linux package zoneminder-git
- Full name and version of OS: Arch linux, up to date
- Browser name and version (if this is an issue with the web interface): Chrome 85

**Describe the bug**

When viewing a recorded event, the "Back" link goes nowhere...

**To Reproduce**
Steps to reproduce the behavior:
1. Click on events (hour, day, week, ...) on the main console page for any given camera
2. Click on any event ID in the resulting list of events
3. Click on the "Back" link (~top left part of the screen, above the Codec selector)
4. Notice how that does *not* bring you back to the list of events.
5. Also notice how the JS console in the browser reports: skins_classic_js_skin-dark-1599142396.js:180 Nothing found to bind to window.history.back();return false; on element 

**Expected behavior**
Clicking on the "Back" link should navigate back to the list of events...

mnoorenberghe, 2020-09-04:
Hey @jstenback, long time, no see :) #3034 should fix this

closed 2020-09-04 by connortechnology

jstenback, 2020-09-04:
Hey @mnoorenberghe! Long time no see, and it's a small world... :) Hope you're doing well in these unprecedented times.

PR #3034 (2020-09-04): Specify a function name for data-on-click to fix the Back/Close button for Events. 
Fixes #3033

My local environment isn't set up to test this but it should work.
commit bd829f8a (2020-09-04): Specify a function name for data-on-click to fix the Back/Close button for Events. Fixes #3033