ZoneMinder/zoneminder #3036: API links broken in reverse-proxy nginx variant

ivdok, 2020-09-10:
**Describe Your Environment**
- ZoneMinder v1.34.20 (TurnKey Linux appliance)

**Describe the bug**
Redirection loops occurs when using Nginx in front of Apache + ZoneMinder, when setting up [https://github.com/pliablepixels/zmNinja/wiki/How-to-configure-APIs-with-nginx](according to zmNinja's) instructions - they are meant for situation when Nginx is the main web server, and not a reverse proxy like in my case.

**To Reproduce**
Steps to reproduce the behavior:
1. Use [https://0paste.com/85247](this) config
2. Open https://zm.example.com/zm/api/host/getversion.json in browser and witness 500 Internal Server Error.
3. See following in Nginx logs: 
```
2020/09/10 03:40:39 [error] 47382#47382: *1 rewrite or internal redirection cycle while processing "/zm/api/index.php", client: 95.79.x.x, server: zm.example.com, request: "GET /zm/api/host/getVersion.json HTTP/2.0", host: "zm.example.com"
```
**Expected behavior**
Working API for zmNinja and SurveillancePro

**Debug Logs**
Unapplicable here.

vajonam, 2020-11-30:
@ivdok 

The problem here is that the index.php is trying to be procecssed by nginx. You can use this config to fix it. The `^~` makes the /zm location take precedence over the php declaration

```
        # ZONEMINDER
        location _^~ /zm_ {
            proxy_pass http://zmserver.localdoaim:8180/zm;
            proxy_connect_timeout    300;
            proxy_send_timeout       300;
            proxy_read_timeout       300;
            send_timeout                     300;
	    proxy_http_version 1.1;             
            proxy_buffering off; # Needed for FIFO raw mode. 

        }
```

This is working well for me.

vajonam, 2020-11-30:
See this. https://stackoverflow.com/questions/29212655/nginx-php-scripts-not-being-called-from-reverse-proxy

vajonam, 2021-06-11:
This can be closed thanks.