ZoneMinder/zoneminder #3046: mariadb shouldn't be a hard requirement in the systemd unit file (multi-server)

dodexahedron, 2020-09-17:
**Describe Your Environment**
- zoneminder-httpd-1.35.6-1.291.20200914gitgba900dd9d.el8.x86_64
- zmrepo
- CentOS 8.2

**Describe the bug**
Multi-server installs will only have the database on one server or on a separate server. The systemd unit file lists mariadb.service as Requires. Thus, if mariadb-server is not installed on the local system, the service won't start.

**To Reproduce**
Steps to reproduce the behavior:
1. Install zoneminder from zmrepo, but don't install mariadb-server.
2. Attempt to start zoneminder

**Expected behavior**
I would think a user would expect it to work out of the box, as the documentation for multi-server doesn't actually say anything about messing with the unit file.

A couple simple solutions:

1. The unit file already lists mariadb.service as After. Change Requires to Wants so that it starts it if it's there, but doesn't fail if it isn't.
2. Provide a multi-server rpm (call it *-multiserver or something) which omits the Requires line from the systemd unit file, and change the normal rpms to have mariadb-server as a dependency.
3. Update the docs to mention needing to dump a modified unit file in /etc/systemd/system/

**Debug Logs**
N/A

knight-of-ni, 2020-09-17:
This is by design, and it absolutely should be a hard requirement. If mariadb were to stop or crash, then zoneminder must stop as well. Back in the sys v init days, this was a real problem. Hence the requires. 

The ZoneMinder documentation for redhat is designed around the least common denominator, which is a single server managed by a user with little to no knowledge of Linux.

The nice thing about ZoneMinder is you can craft it into many different system architectures. This is a bit of double edged sword, however, as we don't have the resources nor desire to document every single niche case.

The multi-server configuration is a special case, is a distro agnostic set of instructions, and is not intended to work out of the box, for any distro. It's something that only an advanced user should try and only after some manual tweaking.

If you want to submit a PR to add a documentation to the multiserver config, then feel free. The systemd service file, however, will not change nor will a multiserver subpackage be created. This is not negotiable.

dodexahedron, 2020-09-17:
Well, then it's a bug anyway, if that's the design intent, because that's not how Requires works. Requires is only checked on service start.
If you want the behavior you're describing, it needs to be using BindsTo, rather than Requires.
mariadb failing will not cause systemd to stop zoneminder, and zoneminder will just crash on its own, with Requires.

With `BindsTo=mariadb.service`, manually stopping mariadb or mariadb failing will cause systemd to stop zoneminder.

knight-of-ni, 2020-09-17:
Interesting point.

The primary design intent is to restart/stop zoneminder automatically if the underlying database service is stopped or restarted. That includes manually by the end user or from a package update or any other condition that does a systemctl stop/restart mariadb. The current zoneminder unit file does this. One can easily test that. Previously, this would happen often enough, with end users posting a page full of errors from their zoneminder log, only to discover a package update, or some other condition, restarted the database service. 

Thanks for pointing out that this won't cover the case where the database service crashes after it was already started. I will update the service file accordingly.

knight-of-ni, 2020-09-17:
The redhat systemd unit file has been updated with commit b3c5a98

dodexahedron, 2020-09-17:
Only BindsTo is necessary, as it includes the behavior of Requires.

knight-of-ni, 2020-12-19:
Multiserver documentation updated with 57b0a96

closed 2020-12-19 by knight-of-ni