ZoneMinder/zoneminder #3048: zmu reporting a different status code (5 now, was 4) after upgrade from 1.34.x to 1.35.6

digitalsmarts, 2020-09-19:
Hi I am now running ZM v1.35.6 which I upgraded to from a v1.34.x version a few weeks ago. I'm using the 'deb https://zmrepo.zoneminder.com/debian/master buster/' repo running on a Debian VM. See below for version info.

root@zoneminder:~# uname -a
Linux zoneminder 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux
root@zoneminder:~# cat /etc/issue
Debian GNU/Linux 10 \n \l

I was using the api call "http://<zm server>/zm/api/monitors/alarm/id:1/command:status.json?token=xxxxx" to get the status of my monitored cameras. This was returning '4' usually and then '2' on alarm/motion. This was on versions < v1.35. After upgrading I am now getting a status code of '5'. I've had a friend upgrade from v1.34 to v1.35.6 and he is now getting the same thing. I've tracked back in the api to find that it calls 'zmu -m$id -s' and when I run zmu -m1 -s I now get a returned value of 5.

When I do a zmu -h and look at the -s option for state it does not even list a return value of 5. Only 0 - 4.  So whilst this may not be a bug something has changed and this is not documented in the binaries help output.  I would like to know what the value 5 is and if this is indeed correct.

root@zoneminder:~# uname -a
Linux zoneminder 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux
root@zoneminder:~# zmu -h
zmu <-d device_path> [-v] [function] [-U<username> -P<password>]
zmu <-m monitor_id> [-v] [function] [-U<username> -P<password>]
General options:
  -h, --help           : This screen
  -v, --verbose          : Produce more verbose output
  -l, --list           : List the current status of active (or all with -v) monitors
Options for use with devices:
  -d, --device [device_path]   : Get the current video device settings for [device_path] or all devices
  -V, --version <V4L version>  : Set the Video 4 Linux API version to use for the query, use 1 or 2
  -q, --query          : Query the current settings for the device
Options for use with monitors:
  -m, --monitor <monitor_id>   : Specify which monitor to address, default 1 if absent
  -q, --query          : Query the current settings for the monitor
  **-s, --state          : Output the current monitor state, 0 = idle, 1 = prealarm, 2 = alarm,
                   3 = alert, 4 = tape**
  -B, --brightness [value]     : Output the current brightness, set to value if given
  -C, --contrast [value]     : Output the current contrast, set to value if given
  -H, --hue [value]        : Output the current hue, set to value if given
  -O, --colour [value]       : Output the current colour, set to value if given
  -i, --image [image_index]    : Write captured image to disk as <monitor_name>.jpg, last image captured
                   or specified ring buffer index if given.
  -S, --scale <scale_%%ge>    : With --image specify any scaling (in %%) to be applied to the image
  -t, --timestamp [image_index]  : Output captured image timestamp, last image captured or specified
                   ring buffer index if given
  -R, --read_index         : Output ring buffer read index
  -W, --write_index        : Output ring buffer write index
  -e, --event          : Output last event index
  -f, --fps            : Output last Frames Per Second captured reading
  -z, --zones          : Write last captured image overlaid with zones to <monitor_name>-Zones.jpg
  -a, --alarm          : Force alarm in monitor, this will trigger recording until cancelled with -c
  -n, --noalarm          : Force no alarms in monitor, this will prevent alarms until cancelled with -c
  -c, --cancel           : Cancel a forced alarm/noalarm in monitor, required after being enabled with -a or -n
  -L, --reload           : Signal monitor to reload settings
  -E, --enable           : Enable detection, wake monitor up
  -D, --disable          : Disable detection, put monitor to sleep
  -u, --suspend          : Suspend detection, useful to prevent bogus alarms when panning etc
  -r, --resume           : Resume detection after a suspend
  -U, --username <username>    : When running in authenticated mode the username and
  -P, --password <password>    : password combination of the given user
  -A, --auth <authentication>  : Pass authentication hash string instead of user details
  -T, --token <token>  : Pass JWT token string instead of user details
root@zoneminder:~#

root@zoneminder:~# zmu -m1 -s
5
root@zoneminder:~# zmu -m1 -s
5
root@zoneminder:~# zmu -m2 -s
5
root@zoneminder:~# zmu -m3 -s
5
root@zoneminder:~#

root@zoneminder:~# zmu -l
  Id Func State TrgState    LastImgTim RdIdx WrIdx LastEvt FrmRate
   1    5     5        0 1600482076.49    50    51  269637   15.19
   2    5     5        0 1600482076.81     3     4  269634   15.24
   3    5     5        0 1600482077.14    36    37  269635   15.10
   4    5     5        0 1600482077.40    19    19  269636   15.17
**root@zoneminder:~# zmu -m1 -a
Alarmed event id: 269637**
root@zoneminder:~# zmu -l
  Id Func State TrgState    LastImgTim RdIdx WrIdx LastEvt FrmRate
   **1    5     3        1 1600482093.38     4     4  269637   15.14**
   2    5     5        0 1600482093.69    57    57  269638   15.15
   3    5     5        0 1600482093.96    89    89  269635   15.04
   4    5     5        0 1600482094.34    73    73  269636   15.10
root@zoneminder:~# zmu -l
  Id Func State TrgState    LastImgTim RdIdx WrIdx LastEvt FrmRate
   1    5     3        1 1600482096.78    54    55  269637   15.21
   2    5     5        0 1600482097.09     8     8  269638   15.16
   3    5     5        0 1600482097.44    41    42  269635   15.18
   4    5     5        0 1600482097.73    23    24  269636   15.14
root@zoneminder:~# zmu -m1 -c
root@zoneminder:~# zmu -l
  Id Func State TrgState    LastImgTim RdIdx WrIdx LastEvt FrmRate
   1    5     5        0 1600482104.38    69    69  269637   15.16
   2    5     5        0 1600482104.77    22    23  269638   15.18
   3    5     5        0 1600482105.05    55    56  269639   15.14
   4    5     5        0 1600482105.40    38    39  269636   15.15
root@zoneminder:~#

Hopefully I've provided enough info for someone to be able to reproduce this. Like I mentioned earlier this has been reproduced already by running the zmu commands on a 1.34.x version and then upgrading to 1.35.6 and performing the same command and immediately getting consistent and different response code; 5.

Regards,

Sean

connortechnology, 2020-09-21:
Right... so I added an UNKNOWN value.  So the numbers all moved up 1 which breaks backwards compatability.

I could undo that, and make UNKNOWN be 5.... and I probably will.  However I wonder if instead of a number we should output the word instead.

digitalsmarts, 2020-09-22:
Personally I'm not bothered if its a numeral or string status so long as I know what it is as I use it in integrations. However, if you do switch to using a string status then again this would break backwards compatibility wouldn't it? Thanks for responding and shedding light on the issue @connortechnology.

Regards,

Sean

commit 98aad6ab (2020-09-22): Make status UNKNOWN be -1 so that zmu -s retains backwards compatibility. Fixes #3048
closed 2020-09-22 by connortechnology