ZoneMinder/zoneminder #3052: Time to next event at 50 years

connortechnology, 2020-09-25:
Many people have reported this and I've seen it too. Not sure how to reproduce exactly.

linuxbox01, 2020-09-28:
I'm at a loss as well.  Thought perhaps we did an upgrade wrong, so uninstalled, wiped db, new installation from rpmfusion 1.34.20.  Same issue.....
Installed from docker image 1.34.18, setup four cameras, working perfect for a day.  Only things that wouldn't work was the rewind in the event view, and the speed dropdown didn't work properly, sometimes had to refresh the page if FF, or the dropdown was selected, but that's not a huge deal for us.  Added in 6 more cameras, left for a day, now the issue is showing up all the time....  We can pan ahead in the event view, but we've never been able to click more than 5 times before the problem shows up.  Sometimes on the first click, sometimes after a few,  but it is persistent.  I can give you access to the system if you have any ideas, or if you want me to try something, or provide any other info, let me know.  I can't seem to find anything in the logs that would indicate any problems.

joedaimarg, 2020-10-04:
Same issue encountered here. I'm using the docker version. Worst on the latest update was, I'm getting this screen when trying to rewind an event.

connortechnology, 2020-10-06:
Fixed in 061c88939429d53d0f8565c110290c4aaca989e3

connortechnology, 2020-10-19:
Fixed in released 1.34.22

closed 2020-10-19 by connortechnology