ZoneMinder/zoneminder #3056: ffmpeg deprecated-declarations

nabbi, 2020-10-11:
**Describe Your Environment**
zoneminder 1.34.21
ffmpeg 4.2.4
gcc 9.3.0

**Describe the bug**
Several compiler warnings for deprecated ffmepg functions

**To Reproduce**
-Wdeprecated-declarations

**Debug Logs**
```
zoneminder-1.34.21/src/zm_ffmpeg.cpp:274:28: warning: ‘AVFormatContext::filename’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_ffmpeg.cpp:274:58: warning: ‘AVFormatContext::filename’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_ffmpeg_camera.cpp:465:65: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:171:38: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:184:38: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:195:92: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:205:60: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_rtsp.cpp:407:40: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_rtsp.cpp:424:47: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_sdp.cpp:354:13: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_videostore.cpp:57:35: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]

zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:329:90: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]

zoneminder-1.34.21/src/zm_remote_camera_rtsp.cpp:350:88: warning: ‘int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]

zoneminder-1.34.21/src/zm_ffmpeg.cpp:88:21: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]
zoneminder-1.34.21/src/zm_mpeg.cpp:49:19: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]

```

Thank you

SteveGilvarry, 2020-10-11:
I don’t think anything is broken as they are warnings, ffmpeg itself also
compiles with these warnings. At some point they will remove it, but it can
take a few years.
I will have a look see if I can deal with them as they just annoy me.

nabbi, 2020-10-11:
> I don’t think anything is broken as they are warnings

Agreed. Nothing broken. Possible proactive cleanup opportunities. Thank you

Carbenium, 2021-05-16:
All warnings have been fixed. Since recently (6c73590243064cd2d5ee34373ec5a7e96b62215b), we run our CI builds with `Werror` to make sure we catch things like this more easily.

closed 2021-05-16 by Carbenium